embassy-rs/embassy #9: restructure interrupt handling - discussion

xoviat, 2020-12-29:
Currently, interrupts are structured such that they are owned by one trait. However, it may be the case that interrupts are used by multiple traits independently, and the current project structure exposes interrupts to the user. I propose the following changes:

- A single struct, similar to ownedinterrupt, that allows traits to add and remove handlers to the interrupt. The first add_handler would enable the interrupt and calling remove_handlers would disable it.
- a new method that would simply wake_by_ref the waker on an interrupt, similar to what I have in my pull request with the waker_interrupt macro.

Thoughts?

Dirbaio, 2020-12-29:
If I understood correctly, you're proposing the following:
- embassy registers all interrupt handlers (with `#[interrupt]`)
- The interrupt handler loops over a list of "second-level" interrupt handlers, calling them all.
- There are methods to register/unregister second-level interrupt handlers.

The advantage would be the user can create an Uarte directly without worrying about interrupts and have it Just Work.

I see are a few problems with this approach:
- How's the memory for the list of second-level interrupt handlers managed?
    - Fixed capacity? choosing it is problematic: either the lib does it which won't work for all users, or the user has to configure it somehow, which defeats the point of having the user not worry about interrupts.
    - dynamically allocated, with eg a Vec? this would need `alloc`, and one of the project's goals is to not require `alloc`.
- The user might want to use some interrupts with embassy and others directly with `#[interrupt]`. If embassy registers all interrupt handlers, this is no longer possible. OwnedInterrupt is designed so that the handler is only registered for the interrupts the user does `take!()` to avoid this.

Dirbaio, 2020-12-29:
> it may be the case that interrupts are used by multiple traits independently

Do you have an example of that? In nrf52 I haven't seen one (1 peripheral = 1 interrupt). If there are such cases on stm32, couldn't that be handled in the embassy-stm32 crate? I think that's preferable to handling this in `embassy`, as it doesn't impose the extra cost on chips that don't need it, like nrf.

My intent with OwnedInterrupt was to solve the "embassy registers irq handlers whether you want it or not" problem (problem number 2 in my previous comment), and not much more.

Dirbaio, 2020-12-29:
> a new method that would simply wake_by_ref the waker on an interrupt, similar to what I have in my pull request with the waker_interrupt macro.

Regarding the `waker_interrupt`, I had this half-written in your PR, but i'll post it here.

I experimented with this idea a while ago, and came to the conclusion it's not the right abstraction.

One issue is that it makes impossible for a peripheral driver to do something "on its own" in response to an interrupt. If the interrupt fires, the waker is woken, but if user code is no longer `.await`ing a driver method, the driver won't receive any `poll()` call, so it has no chance to run any code.

Examples:
- The nrf's RTC updates internal state on interrupt and if it's time, it fires alarms. The API has no `async` fn at all, it's callback-based.
- The nrf's BufferedUarte may have to split a write operation in 2 hardware write operations. For example, -
   - User calls `write` with 100 bytes, when the ringbuffer pointer is 10 bytes before the end.
   - `write` puts 10 bytes at the end, and the remaining 90 bytes wrap around back to the beginning.
   - `write` starts uart send of the first 10 bytes.
   - `write` returns immediately, since all the user's data has been copied to the ringbuffer.
   - user is satisfied their data is written, has no interest in writing more, so does no more calls to BufferedUarte.
   - The interrupt fires, indicating the first 10 bytes are done being sent.
   - The `waker_interrupt` calls the waker
   - The user's task gets woken, it does its `poll()` things but since it's not awaiting any BufferedUarte method call, BufferedUarte has no chance to run mor code.
   - the remaining 90 bytes never get sent..???

As you can see, the `waker_interrupt` abstraction works only for limited use cases. I believe it's best to have a single abstraction (Signal) that's flexible enough to cover all use cases in all drivers.

Aditionally, the cost of having a static `Signal` and signaling it from an interrupt handler is similar to the `waker_interrupt`. A Signal is essentially a waker + a flag internally.

xoviat, 2020-12-29:
Briefly: after considering these issues, you're probably correct. The main reason why I proposed the call back method is that `waker_interrupt` seemed like a clean abstraction that would unburden the end user from having to worry about interrupts, but I could see the problems with that approach: apparently there is one interrupt on nrf for all of the gpio pins. I thought the multiple call-backs could be a solution to that problem, in the event that there were multiple waker_interrupt calls for a single interrupt. Some of the problems that you pointed out I thought could be worked around, but some of them I had not considered.

closed 2020-12-29 by xoviat

Dirbaio, 2020-12-29:
> apparently there is one interrupt on nrf for all of the gpio pins

Yes. GPIOTE stores the wakers for each pin. When the interrupt fires, it wakes only the wakers whose pins have actually changed.

If we were using a "register multiple wakers to the interrupt" abstraction, all the waiting tasks would be woken, not just the right one.