embassy-rs/embassy #11: using embassy without rtc?

xoviat, 2020-12-30:
I finally got around to testing on my device, but received a panic:

```
panic_halt::panic@0x0800b392 (~\.cargo\registry\src\github.com-1ecc6299db9ec823\panic-halt-0.2.0\src\lib.rs:33)
core::panicking::panic_fmt@0x0800a99e (Unknown Source:92)
embassy::time::now@0x0800a390 (~\.cargo\git\checkouts\embassy-52086c5c548bb7b2\c6cf9b8\embassy\src\time\mod.rs:22)
embassy::time::instant::Instant::now@0x08009ab6 (~\.cargo\git\checkouts\embassy-52086c5c548bb7b2\c6cf9b8\embassy\src\time\instant.rs:19)
embassy::executor::Executor::run@0x08009d66 (~\.cargo\git\checkouts\embassy-52086c5c548bb7b2\c6cf9b8\embassy\src\executor\mod.rs:217)
rust_grinder::__cortex_m_rt_main@0x08002292 (~\workspace\tproject\src\main.rs:111)
rust_grinder::__cortex_m_rt_main_trampoline@0x08002212 (~\workspace\tproject\src\main.rs:102)
```

Is it possible to use embassy without rtc? The Qspi example does not have it enabled. stm32 has it, but I haven't used it yet, so it would take me some time to enable.

Dirbaio, 2020-12-30:
Yes, it is definitely possible. The executor only uses Instant if you initialize it with `new_with_alarm`. If you initialize it with `new` it should work without rtc. (but of course you can't use Instant, Timer or Ticker in that case).

Are you on latest master?

xoviat, 2020-12-30:
merging `master` fixed the issue

closed 2020-12-30 by xoviat