embassy-rs/embassy #19: Can't call `task` by recursion.

rise0chen, 2021-01-13:
for example:
``` rust
#[thread::task]
async fn run() {
    info!("hello");
    thread::spawn(run());
}
```

Dirbaio, 2021-01-13:
Memory for tasks is statically allocated, so the number of tasks that can run has to be known upfront. By default `#[task]` allocates memory for 1 instance of the task. 

You can set a higher limit like this: `#[task(pool_size = 20)]`. Depending on what you're doing, 2 may be enough: if the 1st task has finished by the time the 2nd task spawns the 3rd one.

What's your usecase for recursively spawning tasks? Most of the time you can replace them with loops. For example if you want to do something periodically you can do it like this:

```rust

#[task]
async fn run() {
    loop {
        do_stuff();
        Timer::after(Duration::from_secs(1)).await;
    }
}
```

rise0chen, 2021-01-13:
i see. thanks.

i can add `#[task(pool_size = 20)]` ,if i want to recv data from 20 ble connection.  right?

Dirbaio, 2021-01-13:
Yep! pool_size=20 allows up to 20 instances of the same task to be spawned concurrently. The 21th instance would fail.

rise0chen, 2021-01-13:
so cool. thanks

closed 2021-01-13 by rise0chen