embassy-rs/embassy #23: RTC async does not work on nrf52810

timokroeger, 2021-01-15:
```
    Finished dev [optimized + debuginfo] target(s) in 1.27s
     Running `probe-run --chip nRF52810_xxAA C:\Users\XXX\embassy\target\thumbv7em-none-eabi\debug\rtc_async`
  (HOST) INFO  flashing program (19.04 KiB)
  (HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
0.000000 INFO  Hello World!
└─ rtc_async::__cortex_m_rt_main @ src\bin\rtc_async.rs:41
0.000001 INFO  tick
└─ rtc_async::run2::task::{{closure}} @ src\bin\rtc_async.rs:30
0.000002 INFO  BIG INFREQUENT TICK
└─ rtc_async::run1::task::{{closure}} @ src\bin\rtc_async.rs:22
```

Stuck at this forever.

The `async_raw` runs as expected, so the bug might not actually be in the RTC code.

timokroeger, 2021-01-16:
The `rtc_async` example still works at commit db8b4ca56533eb743f8b00b904581bc99e4c9597.
Further investigating if this broke with the timer queue update.

Dirbaio, 2021-01-16:
Hmm, that's strange. 

Are you setting the `52810` cargo feature? the 52810 only has RTC0, the example uses RTC1.

timokroeger, 2021-01-17:
You can find the exact code I used to run the examples here: https://github.com/akiles/embassy/compare/master...timokroeger:nrf52810-async-timer-issue

RTC0, RTC1 is available for all devices, RTC2 only for the bigger MCUs.
When I change the example to RTC0 I observe the same behavior: Stuck forever in sleep mode

timokroeger, 2021-01-17:
There seems to be a off by one error for the nrf52810:
https://github.com/akiles/embassy/commit/b2f9c07e0f238ec60dde043d540a370bb79f186c
With that patch the RTC example works! I have no idea why it required tough.

timokroeger, 2021-01-17:
The issue occurs because of this line:
`Interrupt::#name as u8` in `OwnedInterrupt`

In the `nrf52810-pac` there is no NFC interrupt and thus there is a entry missing in the enum:
https://docs.rs/nrf52810-pac/0.9.0/nrf52810_pac/enum.Interrupt.html

Now the numerical enum value does not match the actual interrupt number anymore.

PR #26 (2021-01-17): Fix IRQ numbers
The raw enum values are not always equal to the interrupt number
fixes #23
commit 9f9d8089 (2021-01-17): Fix IRQ numbers

The raw enum values are not always equal to the interrupt number
fixes #23
closed 2021-01-17 by Dirbaio