embassy-rs/embassy #30: Add Mutex

eupn, 2021-01-20:
A no_std version of [tokio mutex](https://tokio-rs.github.io/tokio/doc/tokio/sync/struct.Mutex.html), [async-std mutex](https://docs.rs/async-std/1.9.0/async_std/sync/struct.Mutex.html). Similar to [this mutex](https://github.com/rust-embedded-community/async-on-embedded/blob/master/async-embedded/src/unsync/mutex.rs).

huntc, 2021-10-05:
Is this essentially closed given https://github.com/embassy-rs/embassy/blob/master/embassy/src/blocking_mutex/mod.rs?

Dirbaio, 2021-10-05:
No, this issue is about an async mutex, where you can `.lock().await`. The one we have is blocking.

vito-chl, 2021-12-30:
Is there a plan to develop it？

Dirbaio, 2021-12-30:
Yes. 

In the meantime, you can use [futures-intrusive's Mutex](https://docs.rs/futures-intrusive/latest/futures_intrusive/sync/index.html), it works great with embassy.

Also, PRs are welcome :)

lulf, 2022-04-28:
This is fixed and part of embassy now

closed 2022-04-28 by lulf