embassy-rs/embassy #38: Add timers for stm32

xoviat, 2021-01-21:
The current plan is this:

1. Wait until stm32-rs/stm32f4xx-hal#228 is merged.
2. Wait submit a subsequent PR that includes interrupts.
3. Submit a PR to embassy that wraps the HAL and implements the timer interface.

Dirbaio, 2021-01-21:
The stm32's RTC is for human time (y/m/d h:m:s). Embassy's Clock/Alarm traits want a *monotonic* clock, one that always ticks up at a fixed rate and never jump back or forward. Doing so would break the executor.

Maybe Clock/Alarm can be implemented with TIMER peripherals instead? I'm not familiar with the stm32 hardware at all, maybe there's a way to clock a TIMER from the 32khz crystal if any, in which case it'd be low power.

Dirbaio, 2021-02-26:
Done in #49

closed 2021-02-26 by Dirbaio