embassy-rs/embassy #51: I2C traits

eupn, 2021-02-17:
I'm extensively using I2C in my project together with `embassy` and it would be nice to have an I2C interface trait to implement for a particular chip (stm32wb55 in my case).

eupn, 2021-02-26:
I ended up using [async-embedded-traits](https://github.com/Disasm/async-embedded-traits) for this. It might be a good idea to have these traits implemented in `embassy` to have a more coherent ecosystem.

Dirbaio, 2021-02-26:
Interesting!

The main difference I see is they take `&mut self` instead of `Pin<&mut Self>`. Pin is necessary so the driver can own and handle interrupts.

Do you have pointers to `async-embedded-traits` implementations? How do they handle interrupts?

eupn, 2021-02-26:
@Dirbaio I have I2C implemented [here](https://github.com/eupn/embassy/blob/add-stm32wb55/embassy-stm32wb55/src/i2c.rs). Though it seems to work fine, I'm not sure about its soundness.

PR #64 (2021-03-06): add i2c trait
closes #51.
closed 2021-03-06 by Dirbaio