embassy-rs/embassy #52: Add #[embassy::main] attribute

Dirbaio, 2021-02-17:
To ease adoption of async and reduce boilerplate for simple projects, we could have a macro that allows you to simply have an async main. No need for creating executor, setting up the clock and alarm.

This'd be optional of course, you'd still be able to set up executor(s) manually.

```rust
#[embassy::main]
async fn main(spawner: Spawner) {
    // here you can do async stuff directly.

    // if you need more tasks you can spawn 
    spawner.spawn(foo()).unwrap()
}
```

xoviat, 2021-03-03:
How about this?

```rust
#[embassy::main(
    use_hse: 16.mhz(),
    sysclk: 48.mhz(),
    pclk1: 24.mhz(),
)]
async fn main(spawner: Spawner, dp, cp, clocks) {
    // here you can do async stuff directly.

    // if you need more tasks you can spawn 
    spawner.spawn(foo()).unwrap()
}
```

Aside from the specific RTC for the alarm, the clocks seem to be the sole configuraton. And we can jus tpass them directly to the clocks configurator.

eupn, 2021-03-05:
@Dirbaio @xoviat 

What about separate macros for clock and main?

Roughly,

```rust
// The type hal::Rcc might be any type that will be passed to main() as clocks
#[embassy::clocks]
fn setup_clocks(dp: ..., cp: ...) -> hal::Rcc {
   // Make initial configuration of clocks
   dp.RCC.constrain(Config { ... })
}

#[embassy::main]
async fn main(spawner: Spawner, dp: ..., cp: ..., clocks: hal::Rcc) {
    // here you can do async stuff directly.

    // if you need more tasks you can spawn 
    spawner.spawn(foo()).unwrap()
}
```

An expansion of these macros would look like this:

```rust
fn __setup_clocks(cp, dp) -> hal::Rcc {
  // Make initial configuration of clocks
  dp.RCC.constrain(Config { ... })
}

async fn __main(spawner: Spawner, dp: ..., cp: ..., clocks: hal::Rcc) {
    // here you can do async stuff directly.

    // if you need more tasks you can spawn 
    spawner.spawn(foo()).unwrap()
}

#[entry]
fn main() {
   let cp = ...;
   let dp = ...;
   let clocks = __setup_clocks(cp, dp);

   // *setup executor*
   let spawner = ...;

   __main(spawner, cp, dp, clocks);
}
```

xoviat, 2021-03-24:
One alternative that I considered was to split the function as such:

```rust
#[embassy:main]
async fn main() {
    let dp = stm32::Peripherals::take().unwrap();
    let cp = cortex_m::peripheral::Peripherals::take().unwrap();

    let rcc = dp.RCC.constrain();

    let clocks = rcc
        .cfgr
        .use_hse(16.mhz())
        .sysclk(48.mhz())
        .pclk1(24.mhz())
        .freeze();

    let spawner = boostrap_embassy!(clocks);

    unwrap!(spawner.spawn(run1()));
    unwrap!(spawner.spawn(run2()));
}
```

xoviat, 2021-03-26:
```rust
#[embassy:main(use_hse=48.mhz())]
async fn main() {
    let (dp, cp, rcc, clocks, interrupts) = embassy::Peripherals::take().unwrap();

   embassy::spawn(task);
}
```

closed 2021-03-29 by xoviat