embassy-rs/embassy #60: Documentation!

Dirbaio, 2021-02-28:


eupn, 2021-03-07:
This includes more examples for stuff such as [`Portal`](https://github.com/akiles/embassy/blob/master/embassy/src/util/portal.rs), [`Signal`](https://github.com/akiles/embassy/blob/master/embassy/src/util/signal.rs), and other built-in primitives like [`InterruptFuture`](https://github.com/akiles/embassy/blob/67a6c4f4692db140f9aa27a920266d70857c06f6/embassy/src/util/signal.rs#L83).

xoviat, 2021-03-21:
Started on this with the [wiki](https://github.com/embassy-rs/embassy/wiki). It's not great for now, but can allow more contributions of quick information, which we can eventually use to create better documentation.

theunkn0wn1, 2021-03-22:
Started working on some in-source documentation such that `cargo _nightly doc` produces meaningful output.

huntc, 2021-10-05:
What sort of documentation should we aspire to here? Perhaps we could enhance the description for clarity.

theunkn0wn1, 2021-10-06:
Embassy's public interface must be documented; meaning anything this crate exports that a user may need, should have usable documentation beyond what `cargo doc` spits out by default. While most things on the public interface have been descriptively named, usage is still quite a confusing aspect. 

Optimally, every function should include a description of what its used for, with what other components, and an example of how to use it.

lulf, 2025-05-26:
We've released all(?) crates, they have documented public apis, and there is a book with growing content. Closing this, and let's create issues for more specific needs going forward.

closed 2025-05-26 by lulf