embassy-rs/embassy #80: Which Mutes should I choose ?

rise0chen, 2021-03-17:
``` Rust
static UART: Once<Mutex<Uarte<pac::UARTE0>>> = Once::new();

#[thread::task(pool_size = 20)]
async fn serial_send(buf: Vec<u8>) {
    let uart = UART.get().unwrap();
    let _err = uart.lock().send(&buf).await;
}
#[thread::task]
async fn serial_recv() {
    let mut buf = [0; 1];
    loop {
        let uart = UART.get().unwrap();
        let _err = uart.lock().receive(&mut buf).await;
    }
}

fn send(buf: &[u8]) {
    thread::spawn(serial_send(Vec::from(buf)));
}

pub fn start(uart: Uarte<pac::UARTE0>) {
    let _err = UART.call_once(|| Mutex::new(uart));

    send(&[1,2,3,4,5]);
    thread::spawn(serial_recv());
}
```

rise0chen, 2021-03-17:
execution flow：
1. `uart.lock().send(&buf).await` on `async fn serial_send`
2. Pending, execute other tasks.
3. `uart.lock().receive(&mut buf).await` on `async fn serial_recv`
4. always locked

xoviat, 2021-03-17:
Not sure what you're talking about. What MCU are you using?

rise0chen, 2021-03-20:
It's solved. I replace `spin` by `futures-intrusive`.

closed 2021-03-20 by rise0chen