embassy-rs/embassy #101: Uart4 on stm32f446 emits garbage.

theunkn0wn1, 2021-03-21:
MCU: `stm32f446` on a nucleo board.

I attached a logic analyzer as well as a FTDI to the TX/RX of Uart4 to observe the results of the test.

the software being run is [here](https://gist.github.com/theunkn0wn1/8caa30cfa7a6e2e76923cd34b4b0ea06)

## Expected result:
Logic analyzer / remote end receives bytes `[0xf0, 0xf0, 0xf0, 0xf0, 0xf0, 0x1, 0xf0, ...]`

## Actual result:
[image]
```py
In [19]: con.read_all()
Out[19]: b'\x00\xff\x00\xff\x00\x00\xff\x00\xff\x06\x00\x00\xff\x00\xff\x00\xff\x00\xff\x00\x00\xff\x00\xff\x00\xff\x00\xff\x00\x00\xff\x00\x00\x00\xff\x00\x00\xff\x00\xff\x00\xff\x00\xff\x00\xff\x00\xff\x00\xff\x00\xff\x00\xff'
```

xoviat, 2021-03-21:
Note: the reason for this was invalid clock configuration.

theunkn0wn1, 2021-03-21:
Not a library issue.

Post mortem:
the examples provided in this repository assume an external HSE, which the nucleo boards do NOT come pre-installed.
As a recommendation, either put a large NOTICE on the clocks config, or please refrain from using the HSE as not all configurations have it.

Closing as invalid.

closed 2021-03-21 by theunkn0wn1