embassy-rs/embassy #124: remove qei trait and impl

xoviat, 2021-03-29:
I originally thought that the qei trait would be useful, but I now think that polling with delays is a better approach. The main problem with the qei trait is that you can lose time (ticks) when using the interrupt approach. I will leave this open for a few days and remove it if no one objects.

Dirbaio, 2021-03-29:
I object! nrf also has a quadrature decoder, was looking forward to implement it. There might be a way to not lose ticks.

xoviat, 2021-03-29:
Losing ticks is a big problem. Suppose we have a loop:

```rust
loop {
    qei.wait_for_rotate(...).await;
    counter += 1;
    update_stuff();
}
```

The problem is that if the qei has changed again while `update_stuff()` is happening, `wait_for_rotate()` needs to return immediately. And it needs to keep returning immediately until the loop has caught up (preferably, the trait would be modified to return a count).

Dirbaio, 2021-03-29:
Oh, I see the problem. Maybe adding a trait for "get ticks" would help? User could do wait_for_rotate then call get_ticks to get how many ticks were actually rotated, which could be more.

xoviat, 2021-03-29:
The problem is how this is implemented on the hw level. The interrupt fires at the range bounds (0, TIMER_MAX). That means the range of the timer needs to be set to the number of ticks. If this is small (which is my use case), it's not possible to query from hw the number of ticks that it has advanced past the interrupt. That doesn't make impl. impossible, but it does mean that a custom interrupt handler would be required and the impl. would need to be more complex.

xoviat, 2021-03-29:
What changed my mind is not that the impl. to solve this is impossible. It's that it's not worth it vs delay with these challenges.

PR #133 (2021-04-02): remove qei trait
closes #124.
closed 2021-04-02 by xoviat