embassy-rs/embassy #126: Support i2c in peripheral mode

davidlattimore, 2021-03-29:
I'm working on an embedded project that has an stm32g0x1 microcontroller which needs to act as an i2c peripheral (aka slave) on one bus and an i2c controller (aka master) on another bus. It also needs to be a peripheral on a SPI bus.

Do you have any thoughts on how this could / should be implemented in Embassy? Assuming I end up using Embassy, I'm happy to do most of the implementation of this myself, but any guidance would be greatly appreciated.

xoviat, 2021-03-30:
Generally, i2c on stm32 is something that we want to support. You need to implement the RTC and I2C traits for stm32g0xx microcontrollers. 

- See the RTC implementation that works on stm32f4: https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/rtc.rs
- See the draft I2C implementation that works on stm32f4: https://github.com/embassy-rs/embassy/pull/92/files

For the I2C implementation, you can copy [this file], and then adjust it to use `InterruptFuture` for transmission of the address, and replace L263-L270 with DMA, like what's done in the stm32f4 PR. The F4 PR is probably almost correct, but I can't currently test it because I don't use I2C in my project. I'll probably keep is as a reference until someone needs I2C for F4.

xoviat, 2021-03-30:
Also, join https://app.element.io/#/room/#embassy-rs:matrix.org so that we can discuss your questions in chat, and there's not so many questions in this issue.

bobmcwhirter, 2021-07-01:
I think we have i2c now implemented for all STM32 boards, if you wouldn't mind taking a look?

If you find your target board(s) are not supported (or include too many bugs), let us know.

Thanks!

Dirbaio, 2021-07-01:
stm32g0 is I2Cv2 which we do support. What's missing is:

- stm32g0 support itself. 
- I2C slave mode (we only have master).

stm32g0 support shouldn't be hard, just enabling it [here](https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/gen_features.py#L15) and fixing what breaks (most likely lack of RCC).

Dirbaio, 2022-02-13:
embassy-stm32 now fully supports G0.

Tracking slave mode support in #622

closed 2022-02-13 by Dirbaio