embassy-rs/embassy #136: Implement blocking APIs

Dirbaio, 2021-04-10:
We want embassy drivers to be usable with both the async `embassy-traits` and the blocking `embedded_hal::blocking` traits.

Reasons:

1. compatibility with the existing ecosystem of drivers.
2. Better code size / cpu usage if you know you're doing a very short operation. For example, for an SPI Ethernet chip you could do async for packet transfers, but blocking for the very short register accesses.

derekdreery, 2021-05-01:
I asked about the signature for the hal traits for SPI, and it was suggested that we might need to use our own functions rather than implement the trait, as the signature might not fit.

```quote
firefrommoonlight
If you're unable to resolve this, consider implementing your transfer without the EH trait, and impl the trait only for non-DMA. EH is suitable for general use cases, but may not apply here
For example, my primary Usart read API is not implementable in EH, and that's fine.
```

derekdreery, 2021-05-01:
After some more chat, it seems I misunderstood the data sheet, and on nrf the input and output DMA buffers are allowed to coincide. In this case it's possible to just implement the traits.

Dirbaio, 2022-02-13:
This is mostly done nowadays, please file issues for particular missing blocking impls.

closed 2022-02-13 by Dirbaio