embassy-rs/embassy #155: [feature requiest] low power mode for peripherals

derekdreery, 2021-05-01:
It would be convenient if peripherals supported going into low power mode (disconnecting GPIO pins, disabling SPI, etc.). I'm not sure if this should live in embassy or in user code. It can be a bit fiddly to write because you need a way to represent the different states, which gets complicated with the typed state you get on Rust (a disconnected pin is not the same type as an input or output pin, even though in reality it is zero-sized and doesn't even exist in machine code).

derekdreery, 2021-05-01:
The ideal solution would use the type system to check that you don't try to do anything with a powered-down peripheral, with this type state completely erased from the final binary (only existing at compiletime).

Dirbaio, 2021-05-02:
This is already possible by dropping the driver. For example, create a Uart instance passing it `&mut p.UARTE0` and `&mut p.Px_xx`. When the Uart instance is dropped, whatever in-progress rx/tx is stopped, the uarte gets disabled, and the pins are deconfigured. 

You can permanently store the UARTE0 and pins, and only construct the Uart when using it.

This is only implemented for Uarte right now, but the goal is to implement it for all drivers. PRs welcome :)

derekdreery, 2021-05-06:
So I think that along the FlexPin this question is answered. The expectation is that you only create peripherals when you are using them, and so you won't need to store them, just store the pin types. If there are any exceptions to this they will be explicitally supported, like the `FlexPin` in `embassy-nrf::gpio`

closed 2021-05-07 by derekdreery