embassy-rs/embassy #193: Spi FullDuplex read_write confusion

ninjasource, 2021-05-19:
Hi! 

Reading the docs for `embassy-traits::spi::FullDuplex<Word>` I found myself drawing parallels to the `embedded_hal::spi::FullDuplex<Word>` trait which differs in that the former deals with slices and the latter single Words.

For the reader's convenience I'll add the most important bits below (minus all the noise and assuming a Word is a byte):
```rust
// embassy
pub trait FullDuplex {
    fn read(&mut self, data: &mut [u8]) -> Self::ReadFuture;
    fn write(&mut self, data: &[u8]) -> Self::WriteFuture;
    fn read_write(&mut self, read: &mut [u8], write: &[u8]) -> Self::WriteReadFuture;
}

// embedded_hal
pub trait FullDuplex {
    fn read(&mut self) -> Result<u8, Self::Error>;
    fn send(&mut self, byte: u8) -> Result<(), Self::Error>;
}

```
The embassy docs say "Due to how full duplex SPI works each `read` call must be preceded by a `write` call." but the function is called read_write implying a read before a write, not the other way round like you would expect it to be. 

From the docs I assumed that the following code should have the same result which it doesn't:

```rust
let tx: [u8; 1] = [123];
let mut rx = [0; 1];

// I assumed that this:
unwrap!(spi.read_write(&mut rx, &tx).await);

// is the same as this:
unwrap!(spi.write(&tx).await);
unwrap!(spi.read(&mut rx).await);

// but the read_write call doesn't not give the correct results 
```

I see that write and read actually call read_write under the hood so it may have more internal relevance than external relevance. Can I be expected to use read_write to perform a typical SPI write read operation or am I holding it wrong?

Dirbaio, 2021-05-19:
> Due to how full duplex SPI works each read call must be preceded by a write call

That bit of the docs is wrong, sorry! Will fix.

If you want to do full duplex SPI just call `read_write`. It works as you'd expect. During the first 8 clock cycles, write[0] is transmitted on MOSI and read[0] is received on MISO. During the next 8, write[1] and read[1] are, etc.

Dirbaio, 2021-05-19:
```rust

// I assumed that this:
unwrap!(spi.read_write(&mut rx, &tx).await);

// is the same as this:
unwrap!(spi.write(&tx).await);
unwrap!(spi.read(&mut rx).await);
```

These two are not equivalent. The first will read+write at the same time. The second will write (discarding data on MISO), then *after the write is done* will read (sending dummy data on MOSI).

If you're doing some kind of "write address, read value" thing you'll have to add "dummy" space on the read buffer to discard data received while sending the address. Or simply do a write then a read.

ninjasource, 2021-05-19:
> If you're doing some kind of "write address, read value" thing you'll have to add "dummy" space on the read buffer to discard data received while sending the address. Or simply do a write then a read.

Got it, that works. Thanks!

closed 2021-06-03 by Dirbaio