embassy-rs/embassy #206: Guide to add new board support?

windoze, 2021-05-26:
I briefly looked into the source code but failed to find a clear path to add support for new boards (namely Seeedstudio XIAO, which is a tiny SAMD21 based MCU).
Would you like to provide some documents or sketch about adding a new board?

Thanks,
Chen

derekdreery, 2021-06-25:
Until a guide comes along the best thing to do is to take the `lib.rs` file of an existing board support crate and start from there. You'll want to generate a peripheral access crate from an SVD file if at all possible to use rather than direct register manipulation.

TDHolmes, 2021-07-09:
[atsamd-rs](https://github.com/atsamd-rs/atsamd) has PAC crates for SAMD chips already, so should be able to start from there rather than do the SVD generation again

joelsa, 2024-05-24:
@jamesmunns added a section on adding new chips recently https://github.com/embassy-rs/embassy/pull/2952

closed 2025-05-26 by lulf