embassy-rs/embassy #261: nrf51 support

Liamolucko, 2021-06-29:
Is `embassy-nrf` planned to support nrf51 chips, or just stick to nrf52?

If it is, one issue is nrf51's lack of a UARTE peripheral - it only has plain UART. Would a single API be provided which abstracts over both, or would they have separate APIs?

Dirbaio, 2021-06-29:
Since it's a bit old, I am not planning to work on it myself, but will review and merge PRs adding support.

I suppose the main annoyance will be lack of DMA-enabled UARTE/SPIM/TWIM, yes. Hopefully it's possible to keep the current API by for example storing the slice to transmit in the state, then writing byte by byte in the interrupt on TXDRDY, and waking the task when all bytes have been transmitted...

The executor should Just Work (it already supports thumbv6m) and RTC seems almost identical too, so it shouldn't be too hard to get *something* working :)

Dirbaio, 2024-03-03:
basic work done in https://github.com/embassy-rs/embassy/pull/2469

issues for follow-up work: 

https://github.com/embassy-rs/embassy/issues/2653
https://github.com/embassy-rs/embassy/issues/2654
https://github.com/embassy-rs/embassy/issues/2655

closed 2024-03-03 by Dirbaio