embassy-rs/embassy #267: nrf: Uart error handling

Liamolucko, 2021-07-01:
I noticed that the `UARTE` peripheral has built-in error handling, but `Uarte` just ignores it and always returns `Ok`. Is this intentional?

If not, would an error cause the reception/transmission to halt, or would they continue?

Dirbaio, 2021-08-05:
Errors just set the flags and reception continues:

> An ERROR event will not stop reception. If the error was a parity error, the received byte will still be transferred into Data RAM, and so will following incoming bytes. If there was a framing error (wrong stop bit), that specific byte will NOT be stored into Data RAM, but following incoming bytes will. 

What's your use case for error detection? It is quite weak, it's easy for some corruption to pass the parity check, so if you care about integrity you should use some kind of CRC anyway, at which point the hardware error detection is not that useful anymore...

lfdominguez, 2024-04-17:
What about `BufferedUarte overrun`, this panicked the entire firmware