embassy-rs/embassy #334: embassy_stm32 ADC and DAC drivers don't configure pin correctly

bgamari, 2021-07-31:
To use the ADC and DAC peripherals on external pins, it is expected that the user set the pin's alternative function to `ANALOG`. However, currently neither `embassy_stm32`'s ADC nor DAC drivers do this.

Moreover, it's unclear how the ADC interface as it exists today would efficiently do so given that the it does not take ownership of the pins which it samples.

bgamari, 2021-07-31:
In the case of the ADC this will require interface faces. I propose that the `AdcPin` `impl`s of the existing GPIO pins are dropped and replaced with (following the example of `gpio::Input`):
```
struct AdcInput<'d, T> {
    pub(create) pin: T,
    phantom: PhantomData<&'d mut T>,
}

// generated
impl<'d, T: AdcPin> AdcInput<'d, T> {
    pub fn new(pin: impl Unborrow<Target = T> + 'd) -> Self {
         unborrow!(pin);

        cortex_m::interrupt::free(|_| unsafe {
            let r = pin.block();
            let n = pin.pin() as usize;
            r.moder().modify(|w| w.set_moder(n, vals::Moder::ANALOG));
        });

        Self {
            pin,
            phantom: PhantomData,
        }
    }
}
```

bobmcwhirter, 2021-08-03:
Do the pins not start/default to analog?

bobmcwhirter, 2021-08-03:
We also have `set_as_analog()` on pin to shorten that, which we attempt to do when non-analog peripherals drop their pins, to return them to that mode.

bgamari, 2021-12-29:
@bobmcwhirter, it turns out that nearly all pins *do* start in analog state.

I am fine with relying on `set_as_analog()` to maintain the invariant that unclaimed pins are always returned to analog state. However, this invariant must be documented. There is currently no way someone unfamiliar with the code could know this.

Dirbaio, 2022-01-01:
We shouldn't rely on unclaimed pins being in a particular state, we should consider them undefined. Some don't start in analog mode (I think?) and there could also be bootloaders mucking with pins before we run.

`AdcInput` is a "typestate", the only thing it does is assert the pin is in a certain mode. To be consistent with the way digital pin modes are handled, it probably makes more sense to keep `read` taking the unclaimed pin, and having it set Analog mode.

IfitNaoned, 2023-08-27:
When i run the ADC example on NUCLEO-4L76RG with a potentionmeter like below , the value readed is always the same ( 26 ).

[image]

It's the same bug ?