embassy-rs/embassy #349: net: UB in TcpSocket

Dirbaio, 2021-08-05:
The lifetime transmute here is UB

https://github.com/embassy-rs/embassy/blob/a361050224e561fcd682041981307f9c49293eeb/embassy-net/src/tcp_socket.rs#L143

The explanation is bogus, because even if the TcpSocket doesn't "touch" these bytes, it has a `&mut` to them, which is already UB.

plaes, 2024-03-26:
@Dirbaio  Is this still relevant? Seems like this code was dropped with commit 931a137f8c5a760c2e06c437c98d14eff3e3a587:
* https://github.com/embassy-rs/embassy/pull/752

Dirbaio, 2024-03-26:
ah yep. this was "fixed" by not implementing ReadBuf indeed. Would be cool to be able to implement it but I don't think it's possible to do soundly without changes to smoltcp.

closed 2024-03-26 by Dirbaio