embassy-rs/embassy #400: STM32F1 family support

mryndzionek, 2021-09-18:
Hello,
I would link experiment with `Blue Pill (STM32F103C8)`, but the STM32F1 is not supported yet. I can take a stab at this, but would be good to have some docs regarding the process. So far I know that I need to add `stm32f1` to `stm32-gen-features/src/lib.rs` and run `stm32-gen-features`, but what is the next step? The first error I see is:
```
error[E0432]: unresolved imports `crate::pac::gpio::vals::Afr`, `crate::pac::gpio::vals::Moder`, `crate::pac::gpio::vals::Ot`
  --> /home/mariusz/sources/Rust/embassy/embassy-stm32/src/i2c/v1.rs:12:30
   |
12 | use crate::pac::gpio::vals::{Afr, Moder, Ot};
   |                              ^^^  ^^^^^  ^^ no `Ot` in `gpio::vals`
   |                              |    |
   |                              |    no `Moder` in `gpio::vals`
   |                              |    help: a similar name exists in the module: `Mode`
   |                              no `Afr` in `gpio::vals`

```
Do I need to re-generate `stm32-metapac`?

Dirbaio, 2021-09-18:
stm32f1 has a different GPIO (v1) than all the other stm32's (v2). See [here](https://docs.google.com/spreadsheets/d/1-R-AjYrMLL2_623G-AFN2A9THMf8FFMpFD4Kq-owPmI/edit#gid=0).

stm32-metapac already [has](https://github.com/embassy-rs/stm32-data/blob/main/data/registers/gpio_v1.yaml) support for it, it's missing in embassy.

The steps should be something like:

- Centralizing all stuff that touches the gpio regs to `gpio.rs` (such as the one in the error you quote, should probably be a pub(crate) fn in gpio.rs that then i2c uses)
- Adding `#[cfg(gpio_v1)]`, `#[cfg(gpio_v2)]` in `gpio.rs` so it works on both v1 and v2.

mryndzionek, 2021-09-19:
Okay, I see... So `gpio_v1` and `gpio_v2` are totally incompatible... Does it make sense then to 'pollute' `gpio.rs` with so many `#[cfg(gpio_vx)]`?

mryndzionek, 2021-09-21:
After some analysis it seems that GPIO will be the easy part. Another thing that is missing and seems to be complicated is RCC. Any hints on that? What is a good reference regarding RCC? Datasheets, or some other, existing codebase?

Dirbaio, 2021-09-21:
> Does it make sense then to 'pollute' gpio.rs with so many #[cfg(gpio_vx)]?

Most of the peripherals have a `mod.rs` with stuff common to all versions, then `v1.rs`, `v2.rs`, etc. [example](https://github.com/embassy-rs/embassy/blob/14aa4265db25adb75fdcaf66798a878ae789bf9c/embassy-stm32/src/spi/mod.rs#L3). There's still quite a bit of copypasted code though that can't be moved to `mod.rs` because it depends on *that one tiny difference*.

The alternative is having a single file with #cfg's. This may be worth it if the differences are small to copypaste less, even if it's a bit more ugly. [exti](https://github.com/embassy-rs/embassy/blob/14aa4265db25adb75fdcaf66798a878ae789bf9c/embassy-stm32/src/exti.rs#L23) is the only one doing it for now (because there was way too many slight variations), but it could see some more use.

Both ways are messy code, but the reality we're dealing with here is messy (lots of different chips with different versions) so it's okay. 

> Another thing that is missing and seems to be complicated is RCC. Any hints on that? What is a good reference regarding RCC? Datasheets, or some other, existing codebase?

Yeah RCC is the main pain of adding a new family. The reference manuals available from ST's website document everything. It helps to take the one from the [stm32-rs project](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/src/rcc.rs) as a reference.

For a first version it's OK if it's not complete, it's fine if it starts up and picks up the right frequencies so eg UART has the right baudrate. We can add the extra fancy bits later.

You should join the [Matrix channel](https://matrix.to/#/#embassy-rs:matrix.org) and say hi! We can help you navigate the cursed seas of RCCs :)

mryndzionek, 2021-09-22:
I making progress with RCC. Will probably have more time on the weekend. I've made a [PR](https://github.com/embassy-rs/stm32-data/pull/85) in stm32-data. Would be good to have it merged by then.

mryndzionek, 2021-09-23:
One more thing I would like to get out the way is the dependency on 'raw' GPIO registers in [embassy-stm32/src/i2c/v1.rs](https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/i2c/v1.rs#L74).  Should I change the `I2c` constructor to accept `OutputOpenDrain` and then just [set_as_af](https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/gpio.rs#L302) instead of `configure_pin` [here](https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/i2c/v1.rs#L34)?

Dirbaio, 2021-09-23:
> change the I2c constructor to accept OutputOpenDrain

No. The convention we use is drivers (uart, i2c, spi, etc) accept bare pin singletons (such as `peripehrals::PC4`) and configure them themselves. (It makes drivers easier to use and significantly reduces boilerplate).

Other than that, yes, that GPIO config code should be moved to the `gpio` module, probably as more methods in `sealed::Pin` trait, like the `set_as_af`.

mryndzionek, 2021-09-23:
Okay, so should I add function `set_as_i2c`, or change `set_as_af` to accept a new enum parameter `otype`?

Dirbaio, 2021-09-23:
Perhaps the 2nd is best, so the output type is always set. Currently other drivers that don't want opendrain (uart, spi) just do set_as_af, which might leave the pin in OpenDrain mode if it was so from before. This is probably a bug, always setting the otype would fix it.

mryndzionek, 2021-09-23:
Would it be fair to say that in general for 'tx' pins type should be PushPull, but for 'rx' it should be Floating?

mryndzionek, 2021-10-05:
Initial STM32F1 support was added in #406.

closed 2021-10-05 by mryndzionek