embassy-rs/embassy #404: Proposal: embassy-lora

lulf, 2021-09-24:
Similar to embassy-net, I'd like to propose adding an embassy-lora module with:

*) async lora radio drivers that works with embassy, i.e. using async SPI
*) rust-lorawan (MAC layer) trait implementations for above radios + radios in embassy-stm32

Then, create examples/embassy-lora with examples of using rust-lorawan with the async radio drivers running with embassy. 

Most of the above exists in drogue-device already, but the trait implementations are not yet fully async. I'd like to do the work on the async versions in embassy if people agree with this.

Dirbaio, 2021-09-27:
Sounds great to me! The more features, the better!

I don't think I'll be able to help much, I'm not personally invested in LoRa. I can help with API design/consistency if you want to.

lulf, 2021-09-30:
PR #410

MathiasKoch, 2022-08-18:
Seems like this can be closed?

closed 2022-08-18 by Dirbaio