embassy-rs/embassy #408: stm32: Lower-level DMA channel definition

theunkn0wn1, 2021-09-28:
As per conversation in matrix:

The current DMA channel does not expose lower-level dma information such as the number of transfers remaining.
For some applications, such as `BufferedUart` and `ReadUntilIdle`, where the transfer may be stopped in the middle, knowing how many bytes were transferred is critical.

timokroeger, 2021-09-29:
The *DMA* peripheral explicitly documents the described scenario as supported.

For *BDMA* I found one stumbling block in peripheral docs (checked L0 and G0):

> Stop and abort a channel
>
> If the application does not need any more the channel, this active channel can be disabled by software. The channel is stopped and aborted but the DMA_CNDTRx register content may not correctly reflect the remaining data transfers versus the aborted source and destination buffer/register

theunkn0wn1, 2021-09-29:
> The _DMA_ peripheral explicitly documents the described scenario as supported.
>
> For _BDMA_ I found one stumbling block in peripheral docs (checked L0 and G0):

I do not believe this is a major issue; as long as BDMA's usart supports the idle interrupt.
My plan for the buffered uart is to utilize the idle interrupt to interrupt the DMA transfer and move the memory.
Since the line is idle, the transfer will be stalled and the number should be consistent.