embassy-rs/embassy #432: nrf/radio: implement the radio peripheral

hannes-hochreiner, 2021-10-16:
I am opening this issue to coordinate efforts. I started work on the implementation of the radio peripheral for the nrf. I only have an nrf52810 for testing, so I will put the radio implementation behind the corresponding feature flag. I am targeting an implementation covering sending and receiving of messages using a limited set of features.

I'll raise a PR as soon as I have a working version. If anyone is interested in the work, you can find the code in [my fork of the embassy project](https://github.com/hannes-hochreiner/embassy).

Dirbaio, 2021-10-16:
Interesting! What protocols are you going to implement?

Some prior art on nrf52 radio stuff: [ieee802154](https://github.com/nrf-rs/nrf-hal/blob/master/nrf-hal-common/src/ieee802154.rs), [ESB](https://github.com/thalesfragoso/esb), [BLE](https://github.com/jonas-schievink/rubble).

The issue with RADIO is that protocols often have very tight timing requirements (for example an ACK must be sent within X us after receiving a packet, so the sending node can retry or sleep faster). A "general" low level radio driver that has async "transmit" and "receive" operations is likely to be too slow to implement many protocols: if you do `receive().await` then `transmit(ack).await`, there's a trip through the executor in the middle which can delay arbitrarily long if the user has other tasks spawned in the same executor, which can cause missing the ACK deadline.

Usually you'd do this by making the very time-critical parts interrupt-driven, and offer async methods for "higher level" operations, like "receive an ieee802154 packet" which does the ack for you. The downside is the driver is no longer "protocol agnostic".

hannes-hochreiner, 2021-10-16:
Thank you for the interesting background on the protocol implementation. My plan was to start with generic `receive` and `transmit` methods, so that proprietary protocols can be implemented. Given that I am new to embassy and that there are quite a lot of options for the radio offered by the chip, I think that will be a good milestone.

lulf, 2024-02-28:
Closing as we have a basic implementation now

closed 2024-02-28 by lulf