embassy-rs/embassy #536: Unreliable BufferedUarte read_exact calls

huntc, 2021-12-10:
I'm having an issue receiving 18 bytes via `BufferedUarte` where not all of the bytes transmitted to it by another device are being received. I have used a scope to confirm that the other device is sending the 18 bytes.

I think it is to do with both the BufferedUarte and also how I read from it in my program. I read in two parts. First, we use `read_exact` to read a 5-byte header with the 5th byte indicating the remaining payload length. Then we `read_exact` the remainder payload.

If I change things to just one `read_exact` of 18 bytes instead of the two calls outlined above then all appears well.

At first, I thought my program might be exceeding the idle time logic (20 bit/baudrate timeout implemented via PPI), but disabling that and ensuring that the rx buffer was sized exactly to 18 bytes did not appear to help.

huntc, 2021-12-10:
I was able to reproduce this given a modification to the examples/nrf/buffered_uart.rs.

```rust
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

#[path = "../example_common.rs"]
mod example_common;

use embassy::executor::Spawner;
use embassy::io::{AsyncBufReadExt, AsyncWriteExt};
use embassy_nrf::buffered_uarte::State;
use embassy_nrf::gpio::NoPin;
use embassy_nrf::{buffered_uarte::BufferedUarte, interrupt, uarte, Peripherals};
use example_common::*;
use futures::pin_mut;

#[embassy::main]
async fn main(_spawner: Spawner, p: Peripherals) {
    let mut config = uarte::Config::default();
    config.parity = uarte::Parity::EXCLUDED;
    config.baudrate = uarte::Baudrate::BAUD115200;

    let mut tx_buffer = [0u8; 18];
    let mut rx_buffer = [0u8; 18]; // NOTE if I set this to a larger size then the problem is deferred!!!

    let irq = interrupt::take!(UARTE0_UART0);
    let mut state = State::new();
    let u = BufferedUarte::new(
        &mut state,
        p.UARTE0,
        p.TIMER0,
        p.PPI_CH0,
        p.PPI_CH1,
        irq,
        p.P0_08,
        p.P0_06,
        NoPin,
        NoPin,
        config,
        &mut rx_buffer,
        &mut tx_buffer,
    );
    pin_mut!(u);

    info!("uarte initialized!");

    // unwrap!(u.write_all(b"Hello!\r\n").await);
    // info!("wrote hello in uart!");

    // Simple demo, reading 8-char chunks and echoing them back reversed.
    loop {
        info!("reading...");
        let mut buf = [0u8; 18];
        unwrap!(u.read_exact(&mut buf[..5]).await);
        unwrap!(u.read_exact(&mut buf[5..]).await);
        info!("read done, got {}", buf);

        // Reverse buf
        // for i in 0..4 {
        //     buf.swap(i, 7 - i);
        // }

        // info!("writing...");
        // unwrap!(u.write_all(&buf).await);
        // info!("write done");

        // // Wait until the bytes are actually finished being transmitted
        // unwrap!(u.flush().await);
[... 123 lines cut ...]
56 TRACE   irq_rx: in state receiving
└─ embassy_nrf::buffered_uarte::{impl#6}::on_interrupt @ /Users/huntc/Projects/hacking/embassy/embassy-nrf/src/fmt.rs:112
57 TRACE   irq_tx: in state Idle
└─ embassy_nrf::buffered_uarte::{impl#6}::on_interrupt @ /Users/huntc/Projects/hacking/embassy/embassy-nrf/src/fmt.rs:112
58 TRACE irq: end
└─ embassy_nrf::buffered_uarte::{impl#6}::on_interrupt @ /Users/huntc/Projects/hacking/embassy/embassy-nrf/src/fmt.rs:112
59 TRACE poll_read
└─ embassy_nrf::buffered_uarte::{impl#3}::poll_fill_buf::{closure#0} @ /Users/huntc/Projects/hacking/embassy/embassy-nrf/src/fmt.rs:112
60 TRACE   got 536871144 6
└─ embassy_nrf::buffered_uarte::{impl#3}::poll_fill_buf::{closure#0} @ /Users/huntc/Projects/hacking/embassy/embassy-nrf/src/fmt.rs:112
61 TRACE consume 6
└─ embassy_nrf::buffered_uarte::{impl#3}::consume::{closure#0} @ /Users/huntc/Projects/hacking/embassy/embassy-nrf/src/fmt.rs:112
62 TRACE irq: start
└─ embassy_nrf::buffered_uarte::{impl#6}::on_interrupt @ /Users/huntc/Projects/hacking/embassy/embassy-nrf/src/fmt.rs:112
63 TRACE   irq_rx: in state receiving
└─ embassy_nrf::buffered_uarte::{impl#6}::on_interrupt @ /Users/huntc/Projects/hacking/embassy/embassy-nrf/src/fmt.rs:112
64 TRACE   irq_tx: in state Idle
└─ embassy_nrf::buffered_uarte::{impl#6}::on_interrupt @ /Users/huntc/Projects/hacking/embassy/embassy-nrf/src/fmt.rs:112
65 TRACE irq: end
└─ embassy_nrf::buffered_uarte::{impl#6}::on_interrupt @ /Users/huntc/Projects/hacking/embassy/embassy-nrf/src/fmt.rs:112
66 TRACE poll_read
└─ embassy_nrf::buffered_uarte::{impl#3}::poll_fill_buf::{closure#0} @ /Users/huntc/Projects/hacking/embassy/embassy-nrf/src/fmt.rs:112
67 TRACE   empty
└─ embassy_nrf::buffered_uarte::{impl#3}::poll_fill_buf::{closure#0} @ /Users/huntc/Projects/hacking/embassy/embassy-nrf/src/fmt.rs:112
```

huntc, 2021-12-10:
If we modify the program to:

```rust
        unwrap!(u.read_exact(&mut buf[..18]).await);
        // unwrap!(u.read_exact(&mut buf[5..]).await);
```

...then all is well!

huntc, 2021-12-12:
Further to the matrix chats, it turns out that the functionality here is presently reliable only when hardware flow control is used. This is a problem that we should address at some stage.

Dirbaio, 2021-12-23:
I believe this should be solved by #537

closed 2021-12-23 by Dirbaio