embassy-rs/embassy #549: Will there be a release?

thejpster, 2021-12-18:
Someone has added embassy support to [rp-hal](https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/Cargo.toml) for the Raspberry Silicon RP2040. I'd like to publish a new version of that package, but we can't whilst we point at a git dependency for embassy-traits.

Will there be a release soon? If not, I will have to back out support for embassy to I can `cargo publish`.

Thanks!

Dirbaio, 2021-12-18:
The goal is to make a release eventually, obviously :) 

The `embassy-traits` stuff is quite up in the air since it will probably be replaced by this soon https://github.com/rust-embedded/embedded-hal/pull/285, so we don't want to make a release to have to immediately break it (and break dependents like rp-rs).

For now I'd just comment out the `embassy-trait` dep in crates.io releases. You also might want to expose async functionality via inherent methods instead of trait methods, so they can be used without `embassy-traits`, and on stable.

Dirbaio, 2022-02-13:
Closing, the requested crate `embassy-traits` is now deprecated and almost-deleted. Async traits are now being developed in `embedded-hal`: https://github.com/rust-embedded/embedded-hal/issues/356

closed 2022-02-13 by Dirbaio