embassy-rs/embassy #570: AVR support?

jeremysalwen, 2022-01-09:
I noticed that this project reserved the crate embassy-avr: https://lib.rs/crates/embassy-avr  Does this foretell AVR support in the near future?

wbrickner, 2023-03-25:
Bump. Is there a way to do this? I'd love to use embassy for all my arduino projects. I don't know how to set this up myself unfortunately

cherrydev, 2023-04-13:
I too would like to see this. Unfortunately I'm pretty much a rust beginner so I'm not going to be much help. That said, the documentation says that you can use a 3rd party HAL with embassy and if you look at the esp-rs/esp-hal repo in the esp-hal-common module you'll see how it's implemented.

niclashoyer, 2023-05-28:
I hacked together a working embassy executor and a working time driver with a fixed frequency. Please be aware that this is just a proof of concept, as there are no async driver implementations for other peripherals yet.

Embassy executor: https://github.com/niclashoyer/embassy/blob/avr/embassy-executor/src/arch/avr.rs

Blink/hello world example (using `simavr` and async delays): https://github.com/niclashoyer/embassy-avr-blink

leftger, 2026-10-09:
This looks like it was addressed — AVR support was added.

`embassy-executor` gained an AVR platform in 58d503a77 ("add avr support", Dec 2023); it is now `embassy-executor/src/platform/avr.rs`.

Reopen or comment if you meant a specific peripheral/async driver on AVR rather than executor support.

closed 2026-10-13 by xoviat