embassy-rs/embassy #571: Issue sending/receiving over UART on STM32 boards.

olofwalker, 2022-01-09:
I'm seeing a weird problem when I try to interface two STM32 Nucleo boards using a UART.

- The application consists of a server and a client that sends strings to each other in non-blocking mode.

- I have wired up the boards with common ground and just an RX/TX cable. It's one Nucleo H7432zi board and one F767zi board.

- I'm using the USART2 and DMA. 

When running in debug mode I see sporadic errors, sometimes I see the server receiving the string with a leading zero. Sometimes the client receives nothing.

I have been using the `trace` debug level to check what data that goes in and out, and the receiving side seems to get a bit of garbage, the data looks fine when sending it.

When running in release mode, or in blocking mode, I have not seen these errors.

It feels like some timing error or some baud rate error.

olofwalker, 2022-01-09:
One thing that was discovered after posting this. On the server-side, the buffer used for RX and TX was the same, when using a separate buffer it seems to work reliably in debug mode also.

Dirbaio, 2022-01-09:
- Make sure you're configuring RCC to clock from HSE. The default is HSI, which is less precise which can cause issues if there's differences between the two boards, as UART is asynchronous.
- Can you check with a logic analyzer on the wire, to see if the data looks OK or not?

olofwalker, 2022-01-11:
@Dirbaio I will have a look at how to configure the RCC to clock from HSE.

I have unfortunately no access to a logic analyzer at this point.

closed 2025-01-10 by Dirbaio