embassy-rs/embassy #575: `cortex_m` config issues for clippy and Visual Studio Code

huntc, 2022-01-11:
As of [Use cortex-m only on cortex-m archs ](https://github.com/embassy-rs/embassy/pull/555), the `rustc` `cfg` `cortex_m` to be declared to use the the `interrupt` module . While this does not present an issue during the building of an application that depends on Embassy, a number of problems surface:

1. `cargo clippy` does not work without `--target` being specified (same for `cargo check` and perhaps others
2. VIsual Studio Code and Rust Analyzer (VSC/RA) are unable to resolve macros such as `interrupt::take!`

Please note that if `.cargo/cargo.toml` declares a default target:

```
[build]
target = "thumbv7em-none-eabihf"
```

...then `cargo clippy` et el will be fine. However, VSC/RA are still unable to resolve the aforementioned macros.

Also, I'd prefer to not declare a default build target as I wish to invoke tests in an `app` `nostd` project via `cargo test`.

I have a PR on my project that reproduces the problem: https://github.com/titanclass/embassy-start/pull/4. The [take macro at L18](https://github.com/titanclass/embassy-start/blob/master/client/embedded-app/src/boards/nrf52840_dk.rs) becomes undefined for VSC/RA. Furthermore, `cargo clippy` et al will fail for both the `client` and `server` projects.

huntc, 2022-01-11:
> You can set `"rust-analyzer.cargo.target": "thumbv7em-none-eabihf",` in `.vscode/settings.json`. This way RA works but no default target is set for actual `cargo` invocations.

Thanks but this doesn’t solve the issue I’ve got now with clippy et al ie now having to declare a target. 

Also I already have a default target set up so this setting shouldn’t be necessary, right?

Can we roll back PR 555, at least partially until we can resolve this? Thanks.

huntc, 2022-01-11:
Thanks, but would it be possible to partially role back 555 temporarily? Things were pretty good before and I don’t think it’s the best dev experience to now have to configure RA for every Embassy based project. Wdyt?

Dirbaio, 2022-01-11:
Assuming cortex-m unless told otherwise is fundamentally wrong. There's other architectures, I hope to support risc-v in the future. Also the user could be building for std without setting the std feature (maybe embassy is a dep of a dep, this was the case for me on iOS).

IMO it's reasonable to have to tell RA your target If you want accurate autocomplete. There are more crates that also expose different stuff depending on the target, including cortex-m itself.

huntc, 2022-01-11:
> "rust-analyzer.cargo.target": "thumbv7em-none-eabihf"

Fair enough. I do think that having to configure RA is short of where we need to be though. How about non RA scenarios?

I'd be entirely happy to configure my default build target aka:

```
[build]
target = "thumbv8m.main-none-eabihf"
```

...which has the same effect as configuring RA explicitly. However, then `cargo test` is problematic. Any thoughts on that

Dirbaio, 2022-01-11:
Having non-embedded tests on an embedded firmware crate is super cursed in my experience. I haven't found a non-horrible way to do that. All the stuff I want to test tends to be separate crates though.

huntc, 2022-01-11:
> 

Just to be clear, I've got two crates within my project - an `app`, which is `no_std` and an `embedded-app` which is the one that actually requires an explicit target. They are both within my VS workspace.

huntc, 2022-01-11:
I've now updated my app to be explicit about targets via .cargo/config.toml files. I'm still not getting RA to recognise the target though given that the `take` macro remains unrecognised. Here are my changes for reference: https://github.com/titanclass/embassy-start/pull/4/files.

noppej, 2022-01-11:
@huntc ... are you able to use VSCode multi-root workspace so that your `app` and `embedded-app` each have their own settings?

huntc, 2022-01-12:
> @huntc ... are you able to use VSCode multi-root workspace so that your `app` and `embedded-app` each have their own settings?

Oh, is that something I must enable? Again, I’m looking for non RA specific solutions though if possible. Thanks.

huntc, 2022-01-12:
> 

Just got back to this. I am actually using a VSC multi-root workspace already: https://github.com/titanclass/embassy-start/blob/master/embassy-start.code-workspace. Unfortunately, even if I declare the app and embedded-app paths instead of just client and server, then there's no difference.

huntc, 2022-01-12:
I'm now declaring the target within the workspace file. This feels cumbersome as RA should be able to "understand" what the build targets are for each project now. I'll close the issue here though as it is indeed reasonable that a target needs to be provide in some way. Thanks for the discussion.

closed 2022-01-12 by huntc

huntc, 2022-01-12:
As I follow up, I think there's an issue with RA: https://github.com/rust-analyzer/rust-analyzer/issues/11268

huntc, 2022-01-14:
My conclusion to this: https://github.com/rust-analyzer/rust-analyzer/issues/11268#issuecomment-1012659059