embassy-rs/embassy #584: GPIOs that can be switched between Input and Output

mryndzionek, 2022-01-22:
Is this possible somehow already?

Dirbaio, 2022-01-22:
nrf has `Flex` that can do exactly that, PRs welcome to port it to stm32 :)

mryndzionek, 2022-01-23:
One more unrelated question. On embedded `time-tick-1000hz` applies, right? Does that mean that `Timer::after` resolution is 1ms? I was making measurements and it seems that the actual resolution on 72MHz STM32F1 is ~25-27us. Any hints on why that is? Would it be feasible to have `time-tick-1mhz` on embedded platform like this?

Dirbaio, 2022-01-23:
it's currently set to [32768hz](https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/Cargo.toml#L48), probably for consistency with nRF (its RTC ticks at a fixed 32768hz), but there's no reason it has to be like that. It'd probably Just Work with other tick rates, AFAICT the [prescaler calculation](https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/time_driver.rs#L109) should deal with it fine.

mryndzionek, 2022-01-23:
Okay. Where is the actual frequency being set? All the code you've linked assumed 32768Hz. There needs to be a place somewhere where TIM2, or TIM3 are being configured to this frequency.

mryndzionek, 2022-01-23:
Okay, I think I managed to switch to `time-tick-1mhz`. I can see that TIM3 is 36MHz and then PSC is set to 35. Unfortunately the resolution seems to be even worse than for `time-tick-32768hz`. About 2x worse...

mryndzionek, 2022-01-23:
Okay, I've made some more measurements and it seems that ~24us is the max resolution I can achieve regardless of the TICKS_PER_SECOND settings. This is probably the cost of /yielding/context switching/.

Dirbaio, 2022-01-23:
What do you exactly mean by "precision"?

Context switching does add some overhead, but it's a constant, so you should still get good precision (low jitter), just slightly higher latency than using raw interrupts. Also 72mhz is pretty fast, added latency should be much less than 24us, in some tests I've seen latency of <6us.

Or are you trying to actually run something at 1mhz? that's just 72 cycles per run, doing a trip through the executor for each run is not going to work, no.

mryndzionek, 2022-01-23:
Yes, I should have written latency. 
The simplest code I've tested is:
```rs
    let mut sig = Output::new(p.PA5, Level::High, Speed::VeryHigh);
    loop {
        sig.toggle();
        Timer::after(Duration::from_micros(2)).await;
    }
```
This gets me a square wave with a period of ~50us, so single pass of the loop takes ~25us. It's the same up to the delay of ~25us and then changes to ~50us.

I'm writing a simple one-wire-type protocol driver.  First I need to generate some 1us-accurate waveform using a GPIO in output mode. Then switch to input and measure time between signal edges.
This is just an experiment. I was curious if this was possible using async function instead of directly in ISR.

Dirbaio, 2022-01-23:
Yeah, at these speeds async is not the best option. You'll also get jitter if you have other tasks in the same executor. EIther use oldschool ISRs, or blocking delays. If the onewire transactions are short, blocking inside the async task is not that bad.

closed 2022-02-13 by Dirbaio