embassy-rs/embassy #622: stm32/i2c: slave mode support

Dirbaio, 2022-02-13:


jrmoulton, 2024-01-25:
I'm interested in implementing stm32 i2c slave mode. I see there was a PR opened a while ago in #2118 but that there were some changes that needed to be made and the PR was marked as a draft. Would it be better to use that PR as a starting point or start from scratch and use the rp implementation as a reference?

bsodmike, 2024-03-08:
Any update on this? Cc @Dirbaio

jrmoulton, 2024-03-09:
I've started implementing it but haven't finished. Will probably get back around to it in a few weeks

jrmoulton, 2024-05-07:
@bsodmike I have a draft PR open in #2909 

Not quite finished but will likely be finished soon

barafael, 2024-06-01:
@Dirbaio looks like this is implemented meanwhile and can be closed

bsodmike, 2024-06-01:
Agreed, this issue can be closed as the PR should be merged soon post CI failures

Knaifhogg, 2025-04-09:
@Dirbaio PR work seems stuck waiting for reviewer/integration plan.

nalepamarcin, 2025-06-27:
@jrmoulton managed to merge his PR https://github.com/embassy-rs/embassy/pull/2909, and with it in place I believe there is a slave mode support for STM32 families that are using `i2c_v2`. How's the situation looking like for the v1? I'm trying to wrap my head around of what's where but I'm kind of overwhelmed by the embassy codebase generation process. Is it possible to switch some v1 board, let's say STM32F103, to v2 implementation to test it out? Or is that just simply impossible (for whatever reason) and the only option is to implement slave mode in v1?

HybridChild, 2025-07-12:
I have done some testing on the current v2 slave mode implementation. You can see the full conclusion [here](https://github.com/HybridChild/embassy-stm32_test_i2c_v2_slave_mode)

## TL;DR: Embassy STM32 I2C v2 Slave Implementation Issues

Embassy STM32 I2C v2 slave implementation is fundamentally broken.

- The `listen()` method intentionally doesn't clear the ADDRCF flag to prevent DMA race conditions, but blocking methods expect it already cleared and the transfer pre-configured. This makes all blocking slave operations timeout.
- The blocking methods skip the hardware initialization that `slave_start()` does (setting NBYTES and clearing ADDRCF) and call internal methods directly.
- Write-read transactions fail because the code only detects STOP conditions, not RESTART conditions used in combined transactions.
- The SBC (Slave Byte Control) mode is enabled but the required byte-by-byte protocol isn't implemented.
- Async methods hang or panic with real buffers due to DMA setup issues.

The result is that I2C v2 slave functionality is completely unusable, forcing developers to write custom register-level code or use different HALs. Tested thoroughly on STM32F072 with comprehensive master/slave scenarios.

Knaifhogg, 2025-07-14:
@HybridChild I have a PR fixing some issues I saw with blocking methods, see if it fixes some of your issues: #4316

closed 2026-02-02 by xoviat