embassy-rs/embassy #623: stm32/spi: slave mode support

Dirbaio, 2022-02-13:


vldm, 2022-07-31:
Hi, i want to implement this mode.
I already experimented with it, and the blocking mode working fine. (two stm32f407 discovery boards communicate correctly)
But then i tried to combine master and slave on single board to emulate Double SPI.
Using DMA, and found an issues, with first clock.

<details>
  <summary>Oscilloscope Image of problem</summary>
Master and slave send &[0b10010001]

[image]
  
</details>

The research lead me to the https://github.com/embassy-rs/embassy/blob/a0f1b0ee01d461607660d2d56b5b1bdc57e0d3fb/embassy-stm32/src/spi/mod.rs#L443-L446 - in this lines spi is disabled.
1. Can you clarify is it important to reset spe flag during reconfiguration, i cannot find any info in my manual about this.

1.1. On my case it works without spi enabled flag changing - so this can be solution.

1.2. Other solution would be to set slave select signal only after reseting spi. But in order to make this work we need to change interface of `transfer_inner` to something like:
```
async fn transfer_inner<W: Word>(
        &mut self,
        read: *mut [W],
        write: *const [W],
        mut handle_slave_select: impl FnMut(),
    ) -> Result<(), Error> 
```
In embedded-has-async they propose to move this logic to level of SPI device https://docs.rs/embedded-hal-async/0.1.0-alpha.1/src/embedded_hal_async/spi.rs.html#59

2. The second issue that blocks me from complete implementation is about sending only one byte (spi word?) trough DMA.
For some reason, in this scenario, it tries to send same byte multiple time. If i send two bytes or more - it works fine.

rokadvednt03, 2022-11-27:
async fn transfer_inner<W: Word>(
        &mut self,
        read: *mut [W],
        write: *const [W],
        mut handle_slave_select: impl FnMut(),
    ) -> Result<(), Error>

leftger, 2026-10-09:
This looks like it was addressed — STM32 SPI slave mode landed in 141f826e1 ("feat: stm32 spi driver slave mode", Nov 2025).

Note that there are currently open reports of hangs when using slave mode (#6291, #6295), so if your interest is slave mode in production, those are the ones to watch. This issue was specifically about the API/feature existing.

Reopen or comment if you need slave mode on a specific family that still isn't covered.

closed 2026-10-09 by leftger