embassy-rs/embassy #630: Creating collections of peripherals

TilBlechschmidt, 2022-02-18:
Hi 👋 

I'm in the process of implementing a key-matrix scanner for a keyboard. However, since the types for the `Input` and `Output` structs have generic parameters to assert ownership of the underlying pin, stuffing multiple instances into an array or vector is not as easy as it sounds.

```rust
let columns_left = [
    Input::new(p.P0_30, Pull::Up),
    Input::new(p.P0_06, Pull::Up),
    Input::new(p.P1_13, Pull::Up),
    Input::new(p.P0_28, Pull::Up),
    Input::new(p.P0_03, Pull::Up),
    Input::new(p.P1_11, Pull::Up),
];
```

The type of the array gets derived as `[Input<P0_30>; 6]` which clashes once the second element is added:

```rust
error[E0308]: mismatched types
    |
125 |         Input::new(p.P0_06, Pull::Up),
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected struct `P0_30`, found struct `P0_06`
    |
    = note: expected type `embassy_nrf::gpio::Input<'_, P0_30>`
             found struct `embassy_nrf::gpio::Input<'_, P0_06>`
```

Maybe there is a stupidly obvious solution — I certainly can't think of it right now though 🙈

Dirbaio, 2022-02-18:
You can use `.degrade()` to convert from a concrete type like `P0_30` to `AnyPin`. Now that all types are the same you can make the array:

```rust
let columns_left = [
    Input::new(p.P0_30.degrade(), Pull::Up),
    Input::new(p.P0_06.degrade(), Pull::Up),
    Input::new(p.P1_13.degrade(), Pull::Up),
    Input::new(p.P0_28.degrade(), Pull::Up),
    Input::new(p.P0_03.degrade(), Pull::Up),
    Input::new(p.P1_11.degrade(), Pull::Up),
];
```

TilBlechschmidt, 2022-02-18:
See, I knew there was a stupidly simple answer to the problem 😆
Thanks for the amazingly quick answer!

Is there maybe some documentation somewhere where I could contribute and add this information?

Dirbaio, 2022-02-18:
Of course, feel free to PR expanding doc-comemnts! There's also some  "guide"-style docs here [https://github.com/embassy-rs/embassy/tree/master/docs]

TilBlechschmidt, 2022-02-18:
Ah yes, I've come across the latter before. Although I am a fan of extensive close-to-code docs, so I'll send a PR for those in a couple of hours 🙂

This issue is done though, thanks for the help!

closed 2022-02-18 by TilBlechschmidt