embassy-rs/embassy #631: Overly aggressive easyDMA SPIM buffer location check?

TilBlechschmidt, 2022-02-19:
When using the nRF52840 SPIM, I've noticed that the `spim.read(&mut rx)` method always throws an `Err(DMABufferNotInMemory)` no matter what. This appears to be related to the fact that it calls `async_inner` with `&[]` as the argument for the tx buffer which appears to have a memory location which `slice_in_ram` does not like — though in my mind it should not matter because it is an empty slice anyways.

Using `transfer` in the same way with an inline array as the tx buffer causes the same issue. It can only be remedied by creating a variable which holds the empty array and passing that as the buffer. See the assertions below:
[image]
[image]

Similarly, this appears to also affect the write method when using inline arrays like `spim.write(&[1, 2, 3])` — probably because the compiler dumps it into the static data section and references it from there instead of allocating it at runtime. I'm not sure there is a reasonable fix for that one.

However, the read method *should* work by either allocating a variable inside the method itself or updating the `slice_in_ram` method to ignore zero-length arrays (given that easyDMA has no problem with that; I have no idea).

Dirbaio, 2022-02-19:
Ooh, nice catch about the zero-length slices. Updating `slice_in_ram` to ignore 0-length slices is OK, EasyDMA ignores the pointer in that case. 

For the `spim.write(&[1, 2, 3])` case, it's unfortunately the way it works. The workaround is to add `copy_write`, `copy_transfer` methods that copy through a RAM buffer, like TWIM has. These'll be required for the embedded_hal impls as well, as they must allow any buffer (HAL-independent drivers can't know of the "must be in RAM" restriction). For the inherent method API (no traits) I still think it's nicer to separate `write` vs `copy_write`, so the user can avoid the overhead if possible.

TilBlechschmidt, 2022-02-20:
Would it make sense to conditionally copy the data? So if the user provides a static buffer that resides in flash memory (and thus `slice_in_ram` fails), it copies it into RAM and then sends its, while a buffer that is already in RAM will be used as is. This would provide the best of both worlds.

However, I am not sure whether this should remain as a conscious choice for the user (e.g. two methods `write` and `maybe_copy_write` [name pending]) or just be the default because otherwise you couldn’t use the copy method anyways. The explicit one reminds the user of potential performance penalties while the implicit one “just works” at the cost of an additional conditional (which is probably still cheaper than the copy operation, but more expensive than the probably more common “send from RAM”).

Dirbaio, 2022-02-23:
Yeah, I agree with you about "just works" being better...

How about this?

- `write_from_ram`: assumes the buffer is in RAM, fails if it doesn't
- `write`: checks if buffer is in RAM. If it is, sends it directly. If not, copies through RAM.

(same would apply to TWI, UART)

TilBlechschmidt, 2022-02-23:
I have a couple of open questions remaining:

1. This kind of check only exists for the nRF sub-crate or would that be something to change across the board (have not found any other similar usages, though GitHub search is not happy today for some reason so I might as well ask)?
2. Would that be something you would want to implement or should I send a PR for review?
3. This change would apply to all variants, so we would have the following list of changes, correct?
    - `write`
    - `write_from_ram`
    - `transfer`
    - `transfer_in_ram`
    - `blocking_write`
    - `blocking_write_in_ram`
    - `blocking_transfer`
    - `blocking_transfer_in_ram`
    - Read functions and in-place transfer always require in-memory mutable slices
    - Additionally, the zero-length fix will apply to all `in_ram` variants

TilBlechschmidt, 2022-02-23:
Oh, right my bad. For some reason my mind went to `in_ram` but I agree that `from_ram` makes much more sense. I'll give it a shot and report back when I've got something tangible 👍

commit 13247897 (2022-03-09): Merge #640

640: Skip EasyDMA slice location check for empty slices and copy data if necessary r=Dirbaio a=TilBlechschmidt

As discussed, this PR makes the following changes:
- Ignore pointer location of zero-length slices (fixes #631)
- Change default functions so they copy the tx buffer if it does not reside in RAM
- Introduce new variants for `write`, `transfer`, and their blocking versions which fails instead of copying
- Add documentation about the motivation behind all these variants
[image]

Remaining TODOs:

- [x] Change copying behaviour for other peripherals
    - [x] TWI
    - [x] U [... cut]
PR #640 (2022-03-09): Skip EasyDMA slice location check for empty slices and copy data if necessary
As discussed, this PR makes the following changes:
- Ignore pointer location of zero-length slices (fixes #631)
- Change default functions so they copy the tx buffer if it does not reside in RAM
- Introduce new variants for `write`, `transfer`, and their blocking versions which fails instead of copying
- Add documentation about the motivation behind all these variants
[image]

Remaining TODOs:

- [x] Change copying behaviour for other peripherals
    - [x] TWI
    - [x] UART
- [x] Add module-level documentation regar [... cut]
closed 2022-03-09 by bors[bot]