embassy-rs/embassy #637: SPI write without read

darkwater, 2022-02-23:
We are using a W5500 ethernet controller, which uses a slightly different SPI protocol. Instead of only supporting 1/2/4 byte read and writes, it also supports N-byte read and writes. Likely to make it more efficient to send/receive bigger network packets. For example, sending a packet might look like this:

```
0f bc 04  11 22 33 44 55 66 77 88 99 aa bb cc dd ee ff
01 02 03  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
^^^^^^^^  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 header    15 bytes to write to address 0x0fbc
```

Reading incoming data looks similar, where the master sends 00 bytes and the W5500 returns a byte of packet data for each 00 byte.

If I want to send a packet, and only send a write operation, the operation seems to hang because there are unread bytes in the receive buffer. My current workaround is just reserving a large buffer and doing a `read_write`, and then discarding the buffer which filled up with 00 bytes:

```rust
async fn write_bytes(&mut self, bytes: &[u8]) -> Result<(), SPI::Error> {
    let mut padding = [0; 512];
    self.1.read_write(&mut padding[..bytes.len()], bytes).await
}
```

I was wondering if it'd be possible to add a method (if there isn't one already that I missed) to only write bytes, while ignoring the bytes sent in response.

Dirbaio, 2022-02-23:
There's `.write()`, which should ignore incoming bytes indeed. If it's hanging it's a bug.

Which chip are you using?

Dirbaio, 2022-03-14:
@darkwater could you confirm which chip you're using?

Dirbaio, 2022-03-15:
if it was stm32, there's been a few hang fixes: #663 #664 . Could you check again? Thank you! :)

Dirbaio, 2023-04-30:
Going to close this since all these SPI bugs are fixed now AFAIK. If the issue persists please ping and I'll reopen.

closed 2023-04-30 by Dirbaio