embassy-rs/embassy #651: Writing large amounts of data with nRF52 EasyDMA yields panic

TilBlechschmidt, 2022-03-02:
As mention by @Dirbaio in #640, writing large amounts of data (`n > FORCE_COPY_BUFFER_SIZE  > EASY_DMA_SIZE`) with methods that use EasyDMA (SPIM, UARTE, TWIM) will yield a panic. This is related to the fact that the internal buffers used can not hold the amount of data passed in.

Both issues can be remedied by adding another layer on-top which splits the data into chunks and transmits those individually.

_Originally posted by @Dirbaio in https://github.com/embassy-rs/embassy/issues/640#issuecomment-1049510130_

Dirbaio, 2022-03-02:
Chunking will work for UARTE and SPIM, not for TWIM.

Unfortunately, the TWIM hardware does a repeated-start every time you start a new transfer. This will break communications with chips that don't expect that repeated-start, and doesn't fulfill the embedded-hal contract. Nordic has confirmed it's not possible to transmit multiple buffers without a repeated-start: https://devzone.nordicsemi.com/f/nordic-q-a/66615/nrf52840-twim-how-to-write-multiple-buffers-without-repeated-start-condition

TilBlechschmidt, 2022-03-02:
That begs the question whether there is somewhat common TWIM hardware that needs transmissions that are larger than a couple hundred bytes or if this is a non-issue. I guess that is something time will tell and whether the use-case arrives.

It might make sense to retain a reasonable panic message in the TWIM API then, I'll re-implement it in my PR.