embassy-rs/embassy #658: secure and non-secure nrf9160 features are incompatible

ZoeyR, 2022-03-10:
The `nrf9160-s` and `nrf9160-ns` features in the `embassy-nrf` crate are incompatible. This becomes an issue when you are trying to develop both a secure bootloader, and a non-secure application in the same workspace. Since cargo does feature unification both secure and non-secure features get activated which results in compilation failures. It would probably be best to either make the features additive or split the secure and non-secure features into sub-crates.

Dirbaio, 2022-03-10:
Mixing different targets/features in the same workspace doesn't really work. Don't do that. You can keep multiple crates in the same repo, just don't create a Cargo workspace.

If you use vscode you can specify which crate to open like [this](https://github.com/embassy-rs/embassy/blob/master/.vscode/settings.json#L21). You can also do multiple `.code-workspace` files with different settings so you can open the same dir with different crates at the same time in different windows.

I don't think this is fixable in embassy, because there are more instances of this issue. If we wanted to support mixing targets in the same workspace, all of the following should work as well. I can't see a way to do that. 

- mixing `nrf9160` and `nrf52840` because your board has both chips
- Have main firmware use `gpiote, time-driver` features, and not the bootloader, to save space.

ZoeyR, 2022-03-10:
That explanation makes sense! Thanks for your work on embassy and especially the nrf 9160 hal.

closed 2022-03-10 by ZoeyR