embassy-rs/embassy #757: RP2040: no probe + newbie questions

wbrickner, 2022-05-07:
Hello! I'm just getting started in embedded rust, I've been wanting to try out embassy, so I picked up a RP Pico board.

As far as I can tell `probe-run` does not actually work out of the box with `RP2040` (and therefore neither does embassy), [the explanation](https://github.com/knurling-rs/probe-run/issues/307) is for now over my head a little.

However, `elf2uf2-rs` works for loading the `blinky` example binary, although it is quite annoying, you need to hold a tiny physical button while simultaneously inserting the microUSB connection in order for a programmed board to mount, then flash with a  `elf2uf2-rs` command.

- Is there some way to streamline this process? I was hoping to be able to `cargo run` like in the examples.

Also, I think my mental model of what is going on must be very wrong, so I will just ask my stupid questions:
- Is there a way to configure the RP Pico such that the development experience is similar to the (seemingly better supported) `STM` / `nRF` chips? As in, `cargo run` and you're done? 
- `probe-run` maintainer suggests installing the 'DapperMime firmware'? Why is this required? I thought the binaries produced _were the firmware_, which would then be flashed to the chip?
- If it would not be too much trouble, would you be able to explain where embassy binaries 'fit in'?  I assumed they were complied to raw instructions that are stored in the flash memory to some magic address, and are then loaded upon chip power on. However, I'm getting confused as there is a lot of talk of a pre-installed ROM bootloader that must be worked around (and handles magical things like providing a UF2 storage interface etc, which I don't really want. I'd like to treat it almost like an AVR/Arduino chip that only runs my embassy binary, although this may be naive).

I would be thankful for any help or explanation.

wbrickner, 2022-05-08:
So I've read more about the situation, and have tried the usual approach of:
- RP Pico A to function as CMSIS-DAP probe for RP Pico B (which I think then enables all the normal functionality of `probe-run`)
- `pico-debug` which is supposed to do the same thing, but using core 2 to function as a probe on core 1.

In either of these cases, the `probe_run` invocation fails identically (and identical to an experiment in which all debug wires are disconnected):

```
Running `probe-run --chip RP2040 target/thumbv6m-none-eabi/release/blinky`
Error: Failed to get information for AP ApAddress { dp: Multidrop(11002927), ap: 0 }
```

What does this mean?  Sorry if these questions are actually hyperspecific to my hardware and setup, I'm not sure what I'm doing yet.

wbrickner, 2022-05-08:
Ha! Okay, completely resolved. Can use `cargo run` (`probe-run`) to flash a Pico preloaded with [`pico-debug`](https://github.com/majbthrd/pico-debug), and even read beautiful log statements in console.

What worked:

1. Install `pico-debug` (I do not want to waste a pico board to function as debugger, plus my breadboard / wiring setup sucks):
  - Connect in USB storage mode (hold button _while_ connecting USB cable)
  - Drag `.uf2` file from one of the `pico_debug` releases (`GIMMECACHE` variant recommended), wait for ~instant ejection / reboot of Pico
3. Modify the `runner` field of `.cargo/config.toml` to change the chip name to `RP2040_SELFDEBUG`:
```toml
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-run --chip RP2040_SELFDEBUG"
```
5. flash, run, watch logs: `DEFMT_LOG=info cargo run --bin blinky --release`

lulf, 2022-05-24:
Glad you figured it out! Is it ok to close this issue?

closed 2022-05-24 by wbrickner