embassy-rs/embassy #774: Cannot enable `time` feature of embassy without choosing a time driver

ZoeyR, 2022-05-17:
I am writing a library that needs to make use of `embassy::time::with_timeout`. To have access to this function the `time` feature needs to be enabled. Without enabling a time driver though I get a compile error
```
error[E0425]: cannot find value `TPS` in this scope
  --> /home/zoey/.cargo/git/checkouts/embassy-9312dcb0ed774b29/a39d796/embassy/src/time/mod.rs:77:35
   |
77 | pub const TICKS_PER_SECOND: u64 = TPS;
   |                                   ^^^ not found in this scope
```

Dirbaio, 2022-05-17:
`with_timeout` can't work if you don't have a time driver, how is it supposed to know what time is it?

The error message could be much better, though.

lulf, 2022-05-17:
Maybe you can use the Delay trait from embedded-hal-async? This way you don't need the driver to depend on the embassy time driver.

https://docs.rs/embedded-hal-async/0.1.0-alpha.0/embedded_hal_async/delay/trait.DelayUs.html

ZoeyR, 2022-05-17:
Yeah, the main issue is that a library can't have code that will work with a timeout without the time drivers. It makes sense that there needs to be a time driver in the final build. But the intermediate libraries needed a way to compile.

ZoeyR, 2022-05-17:
> Maybe you can use the Delay trait from embedded-hal-async? This way you don't need the driver to depend on the embassy time driver.
> 
> https://docs.rs/embedded-hal-async/0.1.0-alpha.0/embedded_hal_async/delay/trait.DelayUs.html

I can't use the `Delay` trait directly since the embassy timer doesn't implement it.

Dirbaio, 2022-05-17:
there's a `Delay` struct that does https://docs.embassy.dev/embassy/git/thumbv7em-none-eabihf/time/struct.Delay.html

ZoeyR, 2022-05-17:
:woman_facepalming: my observation skills fail me again.

ZoeyR, 2022-05-17:
Okay, I can implement a timeout in terms of a `Delay` type. Closing this.

closed 2022-05-17 by ZoeyR