embassy-rs/embassy #786: SysTick time driver

willglynn, 2022-05-28:
Arm Cortex processor cores commonly include a [SysTick](https://developer.arm.com/documentation/dui0497/a/cortex-m0-peripherals/optional-system-timer--systick) peripheral.

SysTick offers some significant advantages. It is a 24-bit downcounter, which means it wraps 256 times less frequently than e.g. an STM32 16-bit timer peripheral, which is desirable at high tick rates. Reading the `SYST_CSR` register returns the 24-bit counter value as well as the `COUNTFLAG`, a flag which it implicitly clears, so absolute counter values can be reliably accumulated without needing [two interrupts per cycle](https://github.com/embassy-rs/embassy/blob/d052631118d8cc998696cf353a5ae7f5b28bbd69/embassy-stm32/src/time_driver.rs#L173). Because SysTick is an Arm peripheral, code which uses SysTick is readily shared between devices from different vendors.

SysTick poses some significant challenges. SysTick supports two [clock sources](https://developer.arm.com/documentation/dui0497/a/cortex-m0-peripherals/optional-system-timer--systick/systick-control-and-status-register): either the processor clock or a vendor-defined "reference clock". Neither of these speeds are known at compile time, nor is there a vendor-agnostic way to determine them, so there will need to be a vendor-specific runtime translation from SysTick ticks to Embassy ticks. Additionally, there is no way to trigger an interrupt from any value except 0. Setting an alarm for a specific point in the current cycle therefore requires modifying the counter register while it's counting – a read-modify-write cycle which perturbs the overall tick count – or using SysTick as a free-running timebase plus a separate hardware timer for near-term events.

I intend to take a stab at implementing an SysTick-based [`embassy::time::driver::Driver`](https://docs.embassy.dev/embassy/git/thumbv6m-none-eabi/time/driver/trait.Driver.html) for STM32. My motivating use case a) requires a high tick rate and b) targets a device with zero 32-bit timers (#784), so a 24-bit timer at 64 MHz (SYSCLK) or especially 8 MHz ("reference clock", here HCLK/8) downsampled to 1 MHz ticks would be a win over a 16-bit timer counting 1 MHz directly. The STM32 time driver using a 16-bit timer peripheral triggers at least two interrupts per cycle – every 32.768ms – vs a hypothetical SysTick driver every 262.144ms or 2097.152ms.

willglynn, 2022-05-29:
My initial impressions:

* A SysTick-based central timer is _so_ not worth it for multi-core devices. It would need to be gated for single-core devices. This shouldn't really be a loss because I don't know of any multi-core devices where SysTick is one of the better timers, but it does suggest a loss of generality.
* Setting alarms is not fun. You set the reload register, then write to the counter to trigger a reload. But you also have to read the counter first to see how many ticks are remaining in this cycle, and read how long this cycle was originally to figure out how many ticks have already elapsed, and you have to accumulate that in a 64-bit counter. All this takes a nonzero amount of time, and if you don't want to lose _those_ ticks, you have to have a calibration step where you measure how long this all takes. And of course you're racing the counter too, so if you're close to a reload event you can do something complicated, or you can just spin, wait for an automatic reload, and try to set the alarm again afterwards. All this adds up to program size.
* Mapping SysTick ticks to Embassy ticks is a performance sink. `armv6m` does not have an integer division instruction, it has a library function. I couldn't make a divisor-decided-at-runtime version of `now()` anything resembling fast.

Some ideas for improvements which I may or may not pursue:

* It would be advantageous to know at compile time that SysTick will run at (say) 8 MHz while Embassy ticks will run at 1 MHz. With that knowledge, `now()` could be specialized at compile time to use a bit shift. This would be a lot faster than dividing by an arbitrary quotient determined at runtime.
* SysTick is part of the core so its tick rate will always be closely related to the speed with which instructions are processed. If the code which alters it takes constant time (or constant enough time), SysTick could be stopped, read, altered, and restarted without functionally losing count. This would be a lot simpler and a lot smaller than leaving it running.

Dirbaio, 2022-05-29:
:+1: to having a "portable" time driver that only uses ARM stuff, but ARM isn't making it easy, yeah... :sweat_smile: 

"Calibrating" how much time stopping+restarting systick takes doesn't seem like a solid solution to me. There's lots of factors that could influence it, like flash cache state, being preempted by interrupts...

There's a few alternative options. Taking inspiration on impls for RTIC's `Monotonic` impls (roughly equivalent to embassy's time driver):

- Make systick overflow at a fixed rate. This means if you want 1kHz precision you need the irq to fire at 1kHz which is not great. https://github.com/rtic-rs/systick-monotonic/blob/master/src/lib.rs
- Using systick for alarms *and* DWT CYCCNT for free-running tick count. Using something intended for debug seems a bit ugly, but it looks very promising otherwise: https://github.com/rtic-rs/dwt-systick-monotonic/blob/master/src/lib.rs

hydra, 2022-06-24:
Using SYSTICK and CYCCNT would be fine for many CPUs, just gate it's availability.  

One of the issues I have is that I need *ALL* the timers on an H7, yes, all of them.  So I can't spare one for embassy's time driver...

MarSik, 2022-08-10:
Not having the systick driver is actually a serious gap. There is a whole class of devices where time precision is less important than power consumption. Even a very naive systick (interrupt - stop - get next planned task time - restart) is enough for those, because the biggest contributor to time jitter is the wakeup time from deep sleep. Typically based on RTC or LPTIM (which is not supported by embassy either atm).

diorcety, 2025-06-28:
I have a working systick driver for stm32, available here: https://github.com/diorcety/embassy/commit/0ab7268dce63cb52967c0ecab449ddeac4b390c3
I currently using it as a "standard systick": 1 tick every 1ms, allowing to plan task with 1ms granularity, with maximum of a bit more 24 days in the future.

Sadly is depends on other modifications (I don't want 64 bits stuff in my 32 bits MCU notably for dealing with timestamps)