embassy-rs/embassy #832: STM32F103C8T6 (Blue pill)  flashed using USB-serial.rs example does not connect properly

AntoineMugnier, 2022-06-25:
I cannot make the USB-serial.rs example works on my blue pill (with Windows).
After the MCU has been flashed, I see the rtt logs appearing up to the following lines:

```
...
0.355010 INFO  Connected
└─ usb_serial::____embassy_main_task::{async_fn#0}::{async_block#0} @ bin\usb_serial.rs:79
0.355072 TRACE READ WAITING, buf.len() = 64
└─ embassy_stm32::usb::usb::{impl#9}::read::{async_block#0} @ C:\Git_projects\embassy\embassy-stm32\src\fmt.rs:112
0.355529 TRACE control: accept OK
└─ embassy_stm32::usb::usb::{impl#11}::accept::{async_block#0} @ C:\Git_projects\embassy\embassy-stm32\src\fmt.rs:112
0.355590 TRACE SETUP read waiting
└─ embassy_stm32::usb::usb::{impl#11}::setup::{async_block#0} @ C:\Git_projects\embassy\embassy-stm32\src\fmt.rs:112
5.393249 TRACE usb: resume
└─ embassy_usb::{impl#2}::handle_bus_event @ C:\Git_projects\embassy\embassy-usb\src\fmt.rs:112
5.393280 TRACE usb: suspend
└─ embassy_usb::{impl#2}::handle_bus_event @ C:\Git_projects\embassy\embassy-usb\src\fmt.rs:112
```
On my PC, I can see that the device is recognized as a VIRTUAL COM PORT, and I see its PID and VID.
I wanted to use the serial-com crate for sending packets to the target and observing its answer, but I fails when openning the com port.

I have tried running the Arduino Hello example on my device, which prints continuously the "hello world" message. For this one I can connect to the port and see the messages.

Dirbaio, 2022-06-25:
> fails when openning the com port.

What error do you get? 

Also, can you try connecting with a pre-made terminal program, like PuTTY?

timokroeger, 2022-06-26:
I had to do some workaround in the nRF example to get the USB serial working on windows:
https://github.com/embassy-rs/embassy/blob/master/examples/nrf/src/bin/usb_serial.rs#L50-L55
Does it also fix your issue?

AntoineMugnier, 2022-06-26:
> What error do you get?

"Unable to open connection to com{x}" with putty

In Windows I also get the warning "This peripheral cannot start (Code 10) " in Devices Manager

AntoineMugnier, 2022-06-26:
> I had to do some workaround in the nRF example to get the USB serial working on windows: https://github.com/embassy-rs/embassy/blob/master/examples/nrf/src/bin/usb_serial.rs#L43-L48 Does it also fix your issue?

It solved it, but for me a simpler workaround is to replace the lines you put:
```
    config.device_class = 0xEF;
    config.device_sub_class = 0x02;
    config.device_protocol = 0x01;
    config.composite_with_iads = true;
```
by
```
    config.device_class = 0x02; // USB CDC 
```
However the user should not have to do this on its own. If I'm right the operation of registering the `device_class` in the usb driver should be done when calling `CdcAcmClass::new()`

closed 2025-05-02 by Dirbaio