embassy-rs/embassy #840: Any work toward ADC for the STM32WL chipset?

risingtiger, 2022-07-02:
Is any work being put toward ADC for the STM32WL chipset?

Any pointers or advice as to best practices in implementing this?

Dirbaio, 2022-07-02:
No one is working on it AFAIK.

First step would be to add it to [stm32-data](https://github.com/embassy-rs/stm32-data). If you're lucky it'll be the same version (`adc_vX`) as another supported family, so you can just reuse it. If not, youll have to also add the driver in embassy-stm32.

risingtiger, 2022-07-04:
I see. I'll look into it

leftger, 2026-10-09:
This looks like it was answered — STM32WL is now covered by the ADC driver.

The unified ADC driver (e42a67a2b, "stm32/adc: single unified driver, add HIL test for all chips") handles the `stm32wl` family; see the family handling in `embassy-stm32/src/adc/internal.rs`.

Reopen or comment if a specific WL ADC feature is still missing.

closed 2026-10-09 by leftger