embassy-rs/embassy #841: Allow changing the ticker speed

ctron, 2022-07-03:
I would like to have a loop, which constantly ticks at a certain rate. However, I also want the user to be able to change that rate during runtime.

It looks like I can simply create a new ticker and use that. However, that would also drop the information when the ticker was last ticked. Internally it has the "expires_at" information, and I think that needs to be carried over to the new ticker, as otherwise, which each change of the ticker, that period would start fresh from the beginning.

Ideally there would be a `set_delay(duration: Duration)` function for the ticker. But maybe there could also be access to the "expires at" information.

peter9477, 2022-09-24:
As a possible workaround, could you record Instant::now() each time the ticker fires, and use that to calculate the remaining time you'd need to wait for the next tick before creating the new Ticker with an updated duration?

There are some details that you don't specify that may be relevant to how this works.  For example, if you had a 1000ms Ticker, and since the last tick 700ms had elapsed but then you decide to change the rate to 500ms, what would you expect to happen?  You've already missed the instant when a 500ms tick would have fired following the previous one, so now what? Is it correct to wait the remaining 300ms to tick at the next one (effectively having missed one tick), or would it be better to tick immediately (since you missed one) and then 500ms later?  

Those sorts of details are application-specific, so it might not be right to encapsulate only one approach in the code by allowing an active ticker to have it's duration adjusted at arbitrary points in time.

What might work as a harmless general feature that would support your use case, however, would be if the Ticker.next() future returned either the Instant::now() at which it had fired, or the expires_at() time when it plans to fire next.  That should let you do the appropriate calculation for your own needs, and would probably support other use cases as well.  Right now it just returns () which is of no value.

leftger, 2026-10-09:
This looks like it was addressed, though as a build-time choice rather than a runtime one.

`embassy-time-driver` now exposes the tick rate as cargo features — `tick-hz-1`, `tick-hz-2`, `tick-hz-4`, `tick-hz-8`, `tick-hz-16`, `tick-hz-32`, `tick-hz-64` (and `133` for RP2040 at 133 MHz via `TIMERx` sourced from `SYSCLK`). These came in with the `embassy-time-driver` split (15f94fb0f).

If what you wanted was changing the tick rate *at runtime*, that still isn't a thing — feel free to reopen on that basis.

closed 2026-10-09 by leftger