embassy-rs/embassy #847: Any work towards ADC support in F3 series?

kacper-cholewinski, 2022-07-06:


Dirbaio, 2022-07-06:
same as #840 

> No one is working on it AFAIK.
> 
> First step would be to add it to [stm32-data](https://github.com/embassy-rs/stm32-data). If you're lucky it'll be the same version (`adc_vX`) as another supported family, so you can just reuse it. If not, youll have to also write a new driver for it in embassy-stm32.

kacper-cholewinski, 2022-07-09:
Can you give me more instructions? Because I think I don't understand fully how to achieve this.

Dirbaio, 2022-07-09:
`embassy-stm32` gets all the info about chips from `stm32-data` (registers, addresses, pins, etc).

The first issue is `stm32-data` is missing the registers for F3 ADC. See [here](https://github.com/embassy-rs/stm32-data/blob/main/data/chips/STM32F303VE.json#L330) there's no `registers` field, compare with [F4](https://github.com/embassy-rs/stm32-data/blob/main/data/chips/STM32F411RC.json#L299-L303).

To fix it, follow the steps [here](https://github.com/embassy-rs/stm32-data#adding-support-for-a-new-peripheral) to generate a YAML for the F3 ADC registers. Diff it with the existing YAMLs in [data/registers](https://github.com/embassy-rs/stm32-data/tree/main/data/registers) to check whether it's compatible with one of the existing YAMLs or not.

If it is, you're lucky. Add an entry for it in [perimap](https://github.com/embassy-rs/stm32-data/blob/main/stm32data/__main__.py#L127), then [regenerate](https://github.com/embassy-rs/stm32-data#generating-the-yamls). This will make the chip file get a `registers` field, which will enable one of these [cfgs](https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/adc/mod.rs#L3) which will pull in the driver for the right version.

If not, you'll need to add a new version to `data/registers` following [these](https://github.com/embassy-rs/stm32-data#adding-support-for-a-new-peripheral) steps, and a new driver in `embassy-stm32` for it.

kacper-cholewinski, 2022-07-11:
Ok, thanks for help. I'm trying but I have some problems with "d" script. Is it supposed to work on win10, or UNIX only?
[image]

kacper-cholewinski, 2022-07-11:
Ok, so I've done it on Ubuntu virtual machine, and I got some YAMLs, and a lot of fails. I've tried LPUART and LPUART1. What next?
[image]
[image]
[image]

Dirbaio, 2022-07-11:
The steps are in my previous post and in the stm32-data README.

kacper-cholewinski, 2022-07-11:
Ok, it seems I don't have necessary knowledge to make this process. Thx for help anyway.

Dirbaio, 2022-07-11:
let's keep the issue open even if you're not planning on working on it yourself, it's a missing feature after all.

appel-pie, 2024-09-19:
Looks like most F3 skews working now save for some needing the f3_v2 version implemented? Looks like that is for the f373 skews.

santhushseta, 2024-11-19:
Hi, did you manage to write the driver for the f373

leftger, 2026-10-08:
This has been addressed by the rework in recent weeks. All ADCs should be supported now.

closed 2026-10-08 by leftger