embassy-rs/embassy #861: STM32F4 family missing USB driver implementation

MisterSandman7, 2022-07-12:
Correct me if i'm wrong, but I am unable to implement the USB peripheral on my Nucleo STM32F413ZH, despite having the physical port and [https://github.com/embassy-rs/stm32-data/blob/b13ba26f6f9b7049097e39ccc7e5e246ac023d15/data/chips/STM32F413ZH.json](url) having the USB peripheral present.

I am unable to create a new driver (with `Driver::new()`) and was wondering if I was missing out on something or not.

I used [https://github.com/embassy-rs/embassy/blob/master/examples/stm32f3/src/bin/usb_serial.rs](url) as example code for my code, since the example repo for the F4 family doesn't have one.

Dirbaio, 2022-07-12:
it's WIP: #799

chemicstry, 2023-01-11:
@Dirbaio this can also be closed with #1145

Dirbaio, 2023-01-11:
Thanks! Closed it will be then.

closed 2023-01-11 by Dirbaio