embassy-rs/embassy #881: embedded-nal(-async) for embassy-net

olanod, 2022-07-29:
Starting to dig into the embedded networking story of Rust I find embassy-net as a very interesting option as I prefer the async nature of the embassy stack. However I initially started prototyping something using [embedded-nal-async](https://github.com/rust-embedded-community/embedded-nal) abstractions to then realize there are no implementations. Found [smoltcp-nal](https://crates.io/crates/smoltcp-nal) that seems like a nice and simple alternative but it doesn't yet implement the async versions of the embedded-nal traits. Is it something that would make sense to implement here?

Dirbaio, 2022-07-29:
The `embedded-nal` (nb-based) traits will not be implemented. `nb` is broken by design, `async` is superior to it in every way, the Embassy project will not invest effort in `nb`. 

The current `embedded-nal-async` has a big design mistake: all methods requiring `&mut self` on the stack means you can't read/write/connect on several sockets concurrently from different tasks. I believe this makes the traits unusable in the real world, `embassy-net` won't implement them until this is fixed.

There's a WIP TcpClient trait that would fix it, we will have an impl for it when it's merged: https://github.com/rust-embedded-community/embedded-nal/pull/69

lulf, 2022-08-18:
Embedded-nal-async has been redesigned and implemented in embassy-net in #895

closed 2022-08-18 by lulf