embassy-rs/embassy #886: Implementing nRF52 QSPI frequency selection

TilBlechschmidt, 2022-07-31:
Hello there!
I would love to implement the feature of being able to select the frequency used for QSPI on the nRF52 ([see this comment](https://github.com/embassy-rs/embassy/blob/bd6bab1625d90a2dc2a4b57b40dcfaa9516bf791/embassy-nrf/src/qspi.rs#L20)).

Couple of notes up-front:
- Should we use the same re-exported `Frequency` enum present in the [SPIM module](https://github.com/embassy-rs/embassy/blob/bd6bab1625d90a2dc2a4b57b40dcfaa9516bf791/embassy-nrf/src/spim.rs#L10), re-export it again, or define our own?
- Are there limits on how low the EasyDMA controller can go? Default seems to be 32 MHz and the previously mentioned enum goes down to 125k.
- Implementation sketch
    - Add frequency parameter to initializer
    - Calculate required frequency divider based on that according to [`SCKFREQ_W`](https://docs.rs/nrf52840-pac/latest/nrf52840_pac/qspi/ifconfig1/struct.SCKFREQ_W.html)'s documentation
    - Set the value [here](https://github.com/embassy-rs/embassy/blob/bd6bab1625d90a2dc2a4b57b40dcfaa9516bf791/embassy-nrf/src/qspi.rs#L125)
    - Update the example(s)

Dirbaio, 2022-07-31:
Should we use the same re-exported Frequency enum present in the [SPIM module](https://github.com/embassy-rs/embassy/blob/bd6bab1625d90a2dc2a4b57b40dcfaa9516bf791/embassy-nrf/src/spim.rs#L10), re-export it again, or define our own?

That one's specific to the SPI freq reg, which is different to SPIM, so no.

> Are there limits on how low the EasyDMA controller can go? Default seems to be 32 MHz and the previously mentioned enum goes down to 125k.

[IFCONFIG1.SCKFREQ](https://infocenter.nordicsemi.com/topic/ps_nrf52840/qspi.html?cp=4_0_0_5_18_9_26#register.IFCONFIG1) is 4 bits, so the max value is 15, which gives the lowest possible freq of 2Mhz.

TilBlechschmidt, 2022-07-31:
Would you prefer an enum that has all possible frequencies, just 32/16/8/4/2, or should the divisor logic be exposed directly to the callee?

(I’d vote for the second option with the limited enum, it is in-line with the SPIM and I don’t see much use for all those intermediate frequencies)

Dirbaio, 2022-08-01:
Enum sounds good. As a general rule it's better to expose all the hardware capabilities, so I'd rather have all 16 possible freqs.

Dirbaio, 2022-08-01:
Yes, these TODOs are outdated. Sorry! :)

TilBlechschmidt, 2022-08-01:
Right, I'll remove them in the PR then :)

TilBlechschmidt, 2022-08-01:
Should I add the SCKDELAY to the config as well? Probably makes sense to do that in one go. Though since it is 255 values I'd probably just expose a `u8` and add the documentation from the nRF docs to it (_Value is specified in number of 16 MHz periods (62.5 ns)._)

TilBlechschmidt, 2022-08-01:
Quick question for understanding the deep-power down: By setting the `DeepPowerDownConfig` value to something, does the EasyDMA peripheral automatically switch it off whenever it is idling for that amount of time, or does that only happen after the `Qspi` struct is dropped (as is done [in the example](https://github.com/embassy-rs/embassy/blob/bd6bab1625d90a2dc2a4b57b40dcfaa9516bf791/examples/nrf/src/bin/qspi_lowpower.rs#L71))

Dirbaio, 2022-08-01:
Only on drop.

IMO adding API to enter/exit DPM without dropping is not useful, because DPM reduces power consumption of the flash chip only, on the nRF side the QSPI peripheral keeps consuming power until you disable it (including funny errata workarounds).

commit a9032154 (2022-08-01): Merge #888

888: Extend QSPI config with freq, delay, spi mode, and address mode r=Dirbaio a=TilBlechschmidt

Fixes #886 among some of the TODOs which have been flying around in the source file :)

I'll make one more commit to remove some old commented out code from the file and take some time this evening to verify that it works.
PR #888 (2022-08-01): Extend QSPI config with freq, delay, spi mode, and address mode
Fixes #886 among some of the TODOs which have been flying around in the source file :)

I'll make one more commit to remove some old commented out code from the file and take some time this evening to verify that it works.
closed 2022-08-01 by bors[bot]