embassy-rs/embassy #898: Dynamic port allocation

arturkow2000, 2022-08-10:
Embassy-net currently allocates ports starting with port 1025 (`LOCAL_PORT_MIN` variable), this may conflict with services that use ports higher than 1025, combined with RNG which IIRC is already used for randomizing ports, may cause random failures in apps.
Shouldn't embassy-net allocate only ports from 49152 to 65535 range?

lulf, 2022-08-10:
It depends on which standard you follow, but I think the IANA range as you suggest is nicer and less surprising than starting at 1025.

Dirbaio, 2022-08-10:
For TCP this is very unlikely to cause issues. What identifies a connection is the 4-tuple (src ip, src port, dst ip, dst port), so you can have a client connection with src port 1234 and a server listening on port 1234 just fine.

For UDP this can be a problem, yes. It can still be a problem if you open so many sockets that you get 2 with the same src port by chance though, so changing the port range doesn't really fix the issue.

I think the right fix would be to explicitly check for conflicts when generating the random src port: generate one, if it's already used retry.

Dirbaio, 2026-08-29:
fixed by #6860 

xarxa now allocates ports in the rfc-mandated ephemeral port 49152-65535 range, and it also checks the port is unused before using it, so it won't conflict with user ports even if they're in the ephemeral port range.

closed 2026-08-29 by Dirbaio