embassy-rs/embassy #917: Why is the embassy-executor require std?

Tnze, 2022-08-21:
I add `embassy-executor = "0"` into my Cargo.toml, and compile it through `cargo b -r`.

```
   Compiling embassy-executor v0.0.0
   Compiling embedded-dma v0.1.2
error[E0463]: can't find crate for `std`
  |
  = note: the `riscv32imac-unknown-none-elf` target may not support the standard library
  = note: `std` is required by `embassy_executor` because it does not declare `#![no_std]`
  = help: consider building the standard library from source with `cargo build -Zbuild-std`

For more information about this error, try `rustc --explain E0463`.
error: could not compile `embassy-executor` due to previous error
```

Dirbaio, 2022-08-21:
It doesn't require std.

Check you're not enabling the `std` Cargo feature in it, or in another crate that might be enabling it indirectly.

You can check with `cargo tree --format '{p} {f}'`

Tnze, 2022-09-12:
This is a toolchain's problem.

closed 2022-09-12 by Tnze

eskorzon, 2023-11-18:
I am having a very similar issue. I can compile and run the example blinky on my board when cloning embassy-rs and doing `cargo run --bin blinky --release` from inside the cloned repo. When I add it through cargo in a fresh repo I get problems:
```
$ cargo run
   Compiling embassy-stm32 v0.0.0
   Compiling heapless v0.7.16
   Compiling cortex-m v0.7.7
error[E0463]: can't find crate for `std`
  |
  = note: the `thumbv7em-none-eabihf` target may not support the standard library
  = note: `std` is required by `embassy_stm32` because it does not declare `#![no_std]`
  = help: consider building the standard library from source with `cargo build -Zbuild-std`

For more information about this error, try `rustc --explain E0463`.
error: could not compile `embassy-stm32` (lib) due to previous error
warning: build failed, waiting for other jobs to finish...
```

Here is the output from cargo tree:
```
$ cargo tree
nucleo-32 v0.1.0 (/home/ekorzon/nucleo-32)
├── cortex-m v0.7.7
│   ├── bare-metal v0.2.5
│   │   [build-dependencies]
│   │   └── rustc_version v0.2.3
│   │       └── semver v0.9.0
│   │           └── semver-parser v0.7.0
│   ├── bitfield v0.13.2
│   ├── embedded-hal v0.2.7
│   │   ├── nb v0.1.3
│   │   │   └── nb v1.1.0
│   │   └── void v1.0.2
│   └── volatile-register v0.2.2
│       └── vcell v0.1.3
├── cortex-m-rt v0.7.3
│   └── cortex-m-rt-macros v0.7.0 (proc-macro)
│       ├── proc-macro2 v1.0.69
│       │   └── unicode-ident v1.0.12
│       ├── quote v1.0.33
│       │   └── proc-macro2 v1.0.69 (*)
│       └── syn v1.0.109
│           ├── proc-macro2 v1.0.69 (*)
│           ├── quote v1.0.33 (*)
│           └── unicode-ident v1.0.12
├── defmt v0.3.5
│   ├── bitflags v1.3.2
│   └── defmt-macros v0.3.6 (proc-macro)
│       ├── defmt-parser v0.3.3
│       │   └── thiserror v1.0.50
│       │       └── thiserror-impl v1.0.50 (proc-macro)
│       │           ├── proc-macro2 v1.0.69 (*)
│       │           ├── quote v1.0.33 (*)
│       │           └── syn v2.0.39
│       │               ├── proc-macro2 v1.0.69 (*)
│       │               ├── quote v1.0.33 (*)
│       │               └── unicode-ident v1.0.12
│       ├── proc-macro-error v1.0.4
│       │   ├── proc-macro-error-attr v1.0.4 (proc-macro)
│       │   │   ├── proc-macro2 v1.0.69 (*)
│       │   │   └── quote v1.0.33 (*)
│       │   │   [build-dependencies]
│       │   │   └── version_check v0.9.4
│       │   ├── proc-macro2 v1.0.69 (*)
│       │   ├── quote v1.0.33 (*)
│       │   └── syn v1.0.109 (*)
│       │   [build-dependencies]
│       │   └── version_check v0.9.4
│       ├── proc-macro2 v1.0.69 (*)
│       ├── quote v1.0.33 (*)
│       └── syn v2.0.39 (*)
├── defmt-rtt v0.4.0
│   ├── critical-section v1.1.2
│   └── defmt v0.3.5 (*)
├── embassy-executor v0.3.3
│   ├── critical-section v1.1.2
│   └── embassy-macros v0.2.1 (proc-macro)
│       ├── darling v0.20.3
│       │   ├── darling_core v0.20.3
│       │   │   ├── fnv v1.0.7
│       │   │   ├── ident_case v1.0.1
│       │   │   ├── proc-macro2 v1.0.69 (*)
│       │   │   ├── quote v1.0.33 (*)
│       │   │   ├── strsim v0.10.0
│       │   │   └── syn v2.0.39 (*)
│       │   └── darling_macro v0.20.3 (proc-macro)
│       │       ├── darling_core v0.20.3 (*)
│       │       ├── quote v1.0.33 (*)
│       │       └── syn v2.0.39 (*)
│       ├── proc-macro2 v1.0.69 (*)
│       ├── quote v1.0.33 (*)
│       └── syn v2.0.39 (*)
├── embassy-stm32 v0.0.0
├── embassy-time v0.1.5
│   ├── cfg-if v1.0.0
│   ├── critical-section v1.1.2
│   ├── embedded-hal v0.2.7 (*)
│   ├── futures-util v0.3.29
│   │   ├── futures-core v0.3.29
│   │   ├── futures-task v0.3.29
│   │   ├── pin-project-lite v0.2.13
│   │   └── pin-utils v0.1.0
│   └── heapless v0.7.16
│       ├── hash32 v0.2.1
│       │   └── byteorder v1.5.0
│       └── stable_deref_trait v1.2.0
│       [build-dependencies]
│       └── rustc_version v0.4.0
│           └── semver v1.0.20
└── panic-probe v0.3.1
    └── cortex-m v0.7.7 (*)
```

Here is my Cargo.toml:
```
[package]
name = "nucleo-32"
version = "0.1.0"
edition = "2021"

[dependencies]
cortex-m = "0.7.7"
cortex-m-rt = "0.7.3"
defmt = "0.3.5"
defmt-rtt = "0.4.0"
embassy-executor = "0.3.3"
embassy-stm32 = "0.0.0"
embassy-time = "0.1.5"
panic-probe = "0.3.1"

```

Here is my .cargo/config.toml:

```
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-run --chip STM32L432KCUx"

[build]
target = "thumbv7em-none-eabihf"
```

Here is my rust-toolchain.toml:
```
[toolchain]
channel = "nightly-2023-11-01"
components = [ "rust-src", "llvm-tools" ]
targets = [
    "thumbv7em-none-eabihf",
]
```

Here is my src/main.rs:
```
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

use defmt::*;
use embassy_executor::Spawner;
use embassy_stm32::gpio::{Level, Output, Speed};
use embassy_time::Timer;
use {defmt_rtt as _, panic_probe as _};

#[embassy_executor::task]
async fn blinker(mut led: Output<'static, PB3>, interval: Duration) {
    loop {
        led.set_high();
        Timer::after(interval).await;
        led.set_low();
        Timer::after(interval).await;
    }
}

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let p = embassy_stm32::init(Default::default());
    info!("Hello World!");

    let led = Output::new(p.PB3, Level::High, Speed::Low);
    unwrap!(spawner.spawn(blinker(led, Duration::from_millis(300))));
}
```

The same code works when compiled from inside a cloned repo, but not when done through a fresh repo where embassy is a dependency. Can embassy be used as a dependency or must it always be used as a clone of this repo? Am I missing some piece of this repo without which nothing will compile correctly?